xmm7360/xmm7360-pci #123: ERROR:root:Attach failed again, giving up

dgiaya, 2021-07-23:
This might not necessarily be a driver problem but I'm hoping by posting it here someone who knows more than I can shed some light on the cause of the problem and hopefully help other with the same problem.

First, my setup:
I'm able to compile and load the xmm7360 module.
I'm on a Lenovo P15s which has the Fibrocom L850GL WWAN card
I'm using a Verizon Wireless SIM that I've verified is unlocked with AT+CPIN?
Using APN vzwinternet
AT+CREG? typically returns 3 (denied)
I have inserted the SIM into my phone and verified it works/is activated

I've seen AT+CREG? return 0 in the past, possibly after switching back and forth between my laptop and phone.

The problem:
I don't seem to get very far after running open_xdatachannel.py. The very first time I ran it it produced a lot of output and eventually ended on a message which I unfortunately did not save but something about 'unrecognized character 0x5'
I've not been able to reproduce that. Every single time thereafter, open_xdatachannel.py ends with ERROR:root:Attach failed again, giving up.

I'm not sure what the UtaMsNetAttachReq involves or where I can learn more about it.

I've also tried with the usb-modeswitch driver but ModemManager also fails to bring up the interface.

Any ideas welcome, thanks!

jonpolak, 2021-08-01:
I have the same problem. T14s (AMD chipset) with ubuntu 21.04  using "Freedom Mobile" 

Similar to OP I verified that SIM CARD is working when swapping into my phone.

tgxn, 2021-08-02:
You could try:
- Make sure you remove every trace of the USB driver+mods+etc.
- Make sure WiFi is turned off.
- There is a similar issue [here](https://github.com/xmm7360/xmm7360-pci/issues/27) that might help you.
- The error you're seeing is [pretty common](https://github.com/xmm7360/xmm7360-pci/issues?q=giving+up) you could see if you can find any solutions in these closed issues.

dgiaya, 2021-08-09:
I've removed the USB driver and mods as far as I can tell.

I've tried from a cold boot:
Turn off wifi
make && make load
sudo python3 rpc/open_xdatachannel.py

I recall that the very first time I loaded the driver and tried it I got the error described here https://github.com/xmm7360/xmm7360-pci/issues/122

I've not been able to replicate that. Now, all I get is ERROR:root:Attach failed again, giving up.

Another thing I've noticed is that I can't seem to talk to the device on /dev/ttyXMM0, only on /dev/ttyXMM1 or /dev/ttyXMM2. I see output from the device when I first open /dev/ttyXMM0 but it does not respond to commands.

Below is the output of open_xdadatachannel.py

`RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x3, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x2
FCC lock: state 1 mode 2
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa0204000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x1, 0x1e0, 0x137, 0x0
response: 0x0, 0xffffffff, 0x0, 0x1, 0xe, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0xe, 0x0
ERROR:root:Attach failed again, giving up
`

yummy527, 2023-04-07:
Did you succeed finally?