xmm7360/xmm7360-pci #124: Clarify licensing of driver module

jan-kiszka, 2021-07-27:
The copyright header is ICS (https://opensource.org/licenses/ISC), MODULE_LICENSE() claims "Dual BSD/GPL", and an SPDX-License-Identifer is missing (checkpatch.pl barks) - what is the actually intended license now? I can't fix that via a PR, that needs to come from the main copyright holder and may require confirmation by all others with significant contributions.

tgxn, 2021-08-01:
To clarify the issue:

License for PCI driver is [here](https://github.com/xmm7360/xmm7360-pci/blob/master/xmm7360.c#L12) which looks to be **`ISC`**

Though [here](https://github.com/xmm7360/xmm7360-pci/blob/master/xmm7360.c#L45) we see the `Dual BSD/GPL`

I would assume that the ISC license is correct, but we'd need @abrasive to advise which the driver should have.

tgxn, 2021-08-02:
From sleuthing in #128 I found that it seems that "Development of this driver was supported by genua GmbH" was [added in Jan 2020](https://github.com/xmm7360/xmm7360-pci/commit/a1d65736a8d805012e6cad2b21c6d284764d7ad5).

Pandapip1, 2024-02-23:
Any updates here?

abrasive, 2024-02-24:
Sorry folks, missed this in amongst the flurry of issues. I double checked the original communications with Genua, we had agreed dual GPL/BSD and then ended up with the wrong license text. This should be fixed in a8ff2c6, can you please confirm?

Pandapip1, 2024-02-26:
LGTM