xmm7360/xmm7360-pci #128: Driver review: xmm_dev.init_work unused

jan-kiszka, 2021-08-02:
init_work

https://github.com/xmm7360/xmm7360-pci/blob/1379b07afb093d56ec63ae9d8d6f02876e0bcd81/xmm7360.c#L203

is only initialized

https://github.com/xmm7360/xmm7360-pci/blob/1379b07afb093d56ec63ae9d8d6f02876e0bcd81/xmm7360.c#L1478

and cleaned up

https://github.com/xmm7360/xmm7360-pci/blob/1379b07afb093d56ec63ae9d8d6f02876e0bcd81/xmm7360.c#L1211

but never triggered. What was the plan with it? Avoid blocking during `xmm7360_probe()`?

tgxn, 2021-08-02:
Related to #104

tgxn, 2021-08-02:
Okay, I've had a look and maybe something to do with failed suspend/resume handling: https://github.com/xmm7360/xmm7360-pci/commit/cc9916560209aa97c6fbb587c21a7cc0127aea8e

jan-kiszka, 2021-08-02:
That makes sense, indeed!

jan-kiszka, 2021-08-02:
OK, suspend/resume seems to be a driver-internal issue according to 2066cdb021102fcc73b6cfb029bec5420864f653. Maybe solvable (was concerned it's related to the device state requiring save/restore).

Anyway, closing this as understood.

closed 2021-08-02 by jan-kiszka

tgxn, 2021-08-03:
Sweet, we can work against #1 if we want to look at doing some work on this.
I'm not a kernel developer, so I'm not of much use with this stuff, but I'll happily help out when I can :)