xmm7360/xmm7360-pci #131: Suspected memory leak in driver

ge0rg, 2021-08-04:
I've noticed that my laptop has been heavily thrashing today (to the extent that keyboard interrupts get dropped or reordered), despite not having any processes with large memory use, and 90% of swap being free:

```
              total        used        free      shared  buff/cache   available
Mem:           62Gi        61Gi       504Mi       1.0Mi       474Mi       320Mi
Swap:          59Gi       5.5Gi        54Gi
```

(yes, this is a high-end machine.) Clearly there is more than enough swap space for it not to thrash and there is no clear culprit.

Some more investigation led to the realization that more than half the physical RAM (almost 34GiB) is allocated by the kernel. According to `slabtop`:

```
 Active / Total Objects (% used)    : 48752216 / 49090462 (99.3%)
 Active / Total Slabs (% used)      : 1810886 / 1810886 (100.0%)
 Active / Total Caches (% used)     : 118 / 160 (73.8%)
 Active / Total Size (% used)       : 35272990.02K / 35367179.59K (99.7%)
 Minimum / Average / Maximum Object : 0.01K / 0.72K / 10.32K

  OBJS ACTIVE  USE OBJ SIZE  SLABS OBJ/SLAB CACHE SIZE NAME                   
8189760 8189689  99%    2.00K 512112       16  16387584K kmalloc-2k
8570020 8570020 100%    1.00K 267849       32   8571168K kmalloc-1k
23704320 23704320 100%    0.25K 740836       32   5926688K skbuff_head_cache
3389544 3387463  99%    0.50K 105951       32   1695216K kmalloc-512
266584 266569  99%    4.00K  33323        8   1066336K kmalloc-4k
1888136 1888136 100%    0.50K  59008       32    944128K skbuff_fclone_cache
1073248 1073146  99%    0.25K  33543       32    268344K kmalloc-256
```
The laptop has an uptime of two weeks, in which time I've extensively used the wwan interface (my cable ISP is down and there is a "free unlimited data" package) to forward all of my home internet traffic through the xmm7360. I haven't experienced this sort of memory clogging before that, so this is a rather strong indication. The 6GB of `skbuff_head_cache` are another pointer to the networking code.

Unloading and re-loading the module doesn't free up any memory, and neither is there an anomaly in `lsmod`.

I've had a weird NAT table running full issue a few days ago, which might be related or not:

```
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
[Sun Aug  1 15:42:37 2021] nf_conntrack: nf_conntrack: table full, dropping packet
```

ge0rg, 2021-08-23:
I've recompiled my kernel with [kmemleak](https://www.kernel.org/doc/html/latest/dev-tools/kmemleak.html) now, just to see where the leakage comes from. It looks like there are other places leaking memory as well, but the network stack is the most significant contributor, and I am rather sure it's xmm7360 not properly releasing SKBs after they are sent.

Here's a bunch of leak dumps after running a bit of `mtr` and `iperf` - there is no clear pointer towards xmm7360, but given that SKBs are reached all around, that doesn't surprise me. Here are my top 3 leaks:

5760 occurences of this backtrace:

```
unreferenced object 0xffff9f88cf63d100 (size 224):
  comm "softirq", pid 0, jiffies 4294937480 (age 359.532s)
  hex dump (first 32 bytes):
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00 00 00 a8 88 9f ff ff 00 00 00 00 00 00 00 00  ................
  backtrace:
    [<0000000036b3e479>] __alloc_skb+0x46/0x1f0
    [<00000000f38230b9>] skb_segment+0x354/0xe30
    [<00000000a76d8feb>] tcp_gso_segment+0xe8/0x4f0
    [<0000000089714e93>] inet_gso_segment+0x14d/0x3b0
    [<000000003f0db6c4>] skb_mac_gso_segment+0x9c/0x110
    [<00000000874e3d3c>] __skb_gso_segment+0xb2/0x160
    [<00000000999ec906>] validate_xmit_skb+0x138/0x270
    [<0000000068a30f50>] validate_xmit_skb_list+0x48/0x70
    [<00000000a84eddb4>] sch_direct_xmit+0x1b4/0x340
    [<0000000003fab0d0>] __qdisc_run+0x14c/0x560
    [<000000002992a79f>] __dev_queue_xmit+0x61c/0x9b0
    [<0000000017c16892>] ip_finish_output2+0x197/0x590
    [<0000000075888516>] __ip_queue_xmit+0x180/0x410
    [<000000001357948f>] __tcp_transmit_skb+0xa0e/0xbc0
    [<00000000461ecfb0>] tcp_write_xmit+0x39b/0x11b0
    [<000000007fe36861>] __tcp_push_pending_frames+0x32/0xf0
```

5713 occurences of this one (same as above, but different offset in `__alloc_skb`:
```
unreferenced object 0xffff9f89b53b9000 (size 4096):
  comm "softirq", pid 0, jiffies 4294937469 (age 358.504s)
  hex dump (first 32 bytes):
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  backtrace:
    [<00000000bb803a9f>] __alloc_skb+0x79/0x1f0
    [<00000000f38230b9>] skb_segment+0x354/0xe30
    [<00000000a76d8feb>] tcp_gso_segment+0xe8/0x4f0
    [<0000000089714e93>] inet_gso_segment+0x14d/0x3b0
    [<000000003f0db6c4>] skb_mac_gso_segment+0x9c/0x110
    [<00000000874e3d3c>] __skb_gso_segment+0xb2/0x160
    [<00000000999ec906>] validate_xmit_skb+0x138/0x270
    [<0000000068a30f50>] validate_xmit_skb_list+0x48/0x70
    [<00000000a84eddb4>] sch_direct_xmit+0x1b4/0x340
    [<0000000003fab0d0>] __qdisc_run+0x14c/0x560
    [<000000002992a79f>] __dev_queue_xmit+0x61c/0x9b0
    [<0000000017c16892>] ip_finish_output2+0x197/0x590
    [<0000000075888516>] __ip_queue_xmit+0x180/0x410
    [<000000001357948f>] __tcp_transmit_skb+0xa0e/0xbc0
    [<00000000461ecfb0>] tcp_write_xmit+0x39b/0x11b0
    [<000000007fe36861>] __tcp_push_pending_frames+0x32/0xf0
```

105 occurences of this:
```
unreferenced object 0xffff9f898e624000 (size 512):
  comm "softirq", pid 0, jiffies 4294936431 (age 246.688s)
  hex dump (first 32 bytes):
    01 7a 62 8e 89 9f ff ff 00 2e 1d 00 8a 9f ff ff  .zb.............
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  backtrace:
    [<0000000036b3e479>] __alloc_skb+0x46/0x1f0
    [<00000000da621235>] sk_stream_alloc_skb+0x63/0x2d0
    [<000000003ca647c6>] tcp_write_xmit+0x5d9/0x11b0
    [<000000007fe36861>] __tcp_push_pending_frames+0x32/0xf0
    [<000000005cd49d72>] tcp_rcv_established+0x354/0x680
    [<000000003a941f60>] tcp_v4_do_rcv+0x131/0x1f0
    [<000000001a5d2a5e>] tcp_v4_rcv+0xc2f/0xd80
    [<00000000904d614d>] ip_protocol_deliver_rcu+0x2b/0x1b0
    [<00000000378ac4a6>] ip_local_deliver_finish+0x44/0x50
    [<00000000828ada50>] __netif_receive_skb_one_core+0x87/0xa0
    [<000000008bf37141>] process_backlog+0x96/0x160
    [<00000000bd2c9b53>] net_rx_action+0x145/0x3e0
    [<00000000b64a9e8c>] __do_softirq+0xc5/0x275
    [<00000000ca3c4fa6>] asm_call_irq_on_stack+0x12/0x20
    [<00000000a7b453da>] do_softirq_own_stack+0x37/0x40
    [<000000001f9fe1d2>] irq_exit_rcu+0x8e/0xc0
```

The other leaks found have significantly fewer individual occurences.

alteriks, 2022-01-07:
Recently my router died so I've been using my X1 Carbon Gen6 as temporary replacement.
Unfortunately PCIE version of xmm7360 driver also freezes my machine. I've got 16GB RAM and 16GB swap. I've noticed that the leak is huge - over 1G per hour. My laptop dies in less than 24h after OOM killer has nothing to kill or some deadlock happens.
For me solution is to revert to https://github.com/xmm7360/xmm7360-usb-modeswitch after using it for two days memory stays at the same level.

handsomeliwenhao, 2023-06-01:
have you resolve this porblem?i meet the similar like yours

ge0rg, 2023-06-01:
Sorry to say that I "resolved" it by putting my SIM card into an old android phone.

oblivcheck, 2024-11-11:
same.
i plan to switch to LG830 :(