xmm7360/xmm7360-pci #132: Works in Thinkpad X1 gen7 Ubuntu 21.04 but laptop freezes after a while

pberthier, 2021-08-13:
Device is working and I can use the network.  But after a while browsing (5-10min) some commands (for example "top") return segmentation fault and shortly after the laptop freezes completely.  After reboot everything works again.

Ubuntu 21.04
Kernel 5.11.0-25-generic #27-Ubuntu SMP Fri Jul 9 23:06:29 UTC 2021 x86_64 
Driver code commit b28714b
No SIM PIN
With driver signing

dcalano, 2021-08-17:
Interesting. Worried to try this. I just loaded and ran this and it worked under a brief test. I did not test it for 5 minutes though. I am also running Pop!_OS 21.04 so I wonder if that might make any differences. Will test and report back

pberthier, 2021-09-02:
I've tested it a bit more and I cannot reproduce the crash.  Now running 
Linux 5.11.0-31-generic #33-Ubuntu SMP Wed Aug 11 13:19:04 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

So it might have been an independent problem..
Thanks so much for the driver!!

closed 2021-09-02 by pberthier