xmm7360/xmm7360-pci #136: Issues compiling locally

evlli, 2021-09-08:
when I try to compile the module, make just tells me this.

```
make -C /lib/modules/5.14.1-arch1-1/build M=/home/asra/git/xmm7360-pci modules
make[1]: Entering directory '/usr/lib/modules/5.14.1-arch1-1/build'
  CC [M]  /home/asra/git/xmm7360-pci/xmm7360.o
/home/asra/git/xmm7360-pci/xmm7360.c:1322:23: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |         .write_room = xmm7360_tty_write_room,
      |                       ^~~~~~~~~~~~~~~~~~~~~~
/home/asra/git/xmm7360-pci/xmm7360.c:1322:23: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:271: /home/asra/git/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1851: /home/asra/git/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.14.1-arch1-1/build'
make: *** [Makefile:9: default] Error 2
```

changing the xmm7360_tty_write_rom struct to be unsigned seems to solve the problem

r4770, 2021-09-15:
I have the same issue.. 

Fedora 34 with XanMod kernel 5.14

```bash
[root@fedora xmm7360-pci]# uname -a
Linux fedora 5.14.3-xm1.0e20210902.fc34.x86_64 #1 SMP PREEMPT Tue Sep 14 04:25:37 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

[root@fedora xmm7360-pci]# gcc --version
gcc (GCC) 11.2.1 20210728 (Red Hat 11.2.1-1)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[root@fedora xmm7360-pci]# make
make -C /lib/modules/5.14.3-xm1.0e20210902.fc34.x86_64/build M=/home/raffaele/xmm7360-pci modules
make[1]: ingresso nella directory «/usr/src/kernels/5.14.3-xm1.0e20210902.fc34.x86_64»
  CC [M]  /home/raffaele/xmm7360-pci/xmm7360.o
/home/raffaele/xmm7360-pci/xmm7360.c:1174:11: warning: address of array 'xmm->td_ring' will always evaluate to 'true' [-Wpointer-bool-conversion]
        if (xmm->td_ring) {
        ~~  ~~~~~^~~~~~~
/home/raffaele/xmm7360-pci/xmm7360.c:1322:16: error: incompatible function pointer types initializing 'unsigned int (*)(struct tty_struct *)' with an expression of type 'int (struct tty_struct *)' [-Werror,-Wincompatible-function-pointer-types]
        .write_room = xmm7360_tty_write_room,
                      ^~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
make[2]: *** [scripts/Makefile.build:271: /home/raffaele/xmm7360-pci/xmm7360.o] Errore 1
make[1]: *** [Makefile:1840: /home/raffaele/xmm7360-pci] Errore 2
make[1]: uscita dalla directory «/usr/src/kernels/5.14.3-xm1.0e20210902.fc34.x86_64»
make: *** [Makefile:9: default] Errore 2
```

How do you bypass this?
Thanks

samtech09, 2021-09-16:
Having same issue with kernel 5.14.3 on Arch Linux.

Any fix for that ?

samtech09, 2021-09-16:
> when I try to compile the module, make just tells me this.
> 
> ```
> make -C /lib/modules/5.14.1-arch1-1/build M=/home/asra/git/xmm7360-pci modules
> make[1]: Entering directory '/usr/lib/modules/5.14.1-arch1-1/build'
>   CC [M]  /home/asra/git/xmm7360-pci/xmm7360.o
> /home/asra/git/xmm7360-pci/xmm7360.c:1322:23: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
>  1322 |         .write_room = xmm7360_tty_write_room,
>       |                       ^~~~~~~~~~~~~~~~~~~~~~
> /home/asra/git/xmm7360-pci/xmm7360.c:1322:23: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:271: /home/asra/git/xmm7360-pci/xmm7360.o] Error 1
> make[1]: *** [Makefile:1851: /home/asra/git/xmm7360-pci] Error 2
> make[1]: Leaving directory '/usr/lib/modules/5.14.1-arch1-1/build'
> make: *** [Makefile:9: default] Error 2
> ```
> 
> changing the xmm7360_tty_write_room function to be unsigned seems to solve the problem

yes changing function to unsigned solve the issue. Thanks.

nursoda, 2021-09-16:
I confirm that. Solved the issue for me in Arch using kernel 5.14.3. I created a [PR](https://github.com/xmm7360/xmm7360-pci/pull/139) for that.

tgxn, 2022-02-10:
Hey @evlli @r4770 @samtech09 

Now that these fixes have been merged, Can you please test that the master branch builds and works well with your current kernel versions?

evlli, 2022-03-10:
Looks good to me! `5.16.13-arch1-1` on Thinkpad X395

closed 2022-03-10 by evlli

JoannaAtZipline, 2023-02-21:
Having this issue right now
```
/home/joannasousa/software/xr_usb_serial_common-1a/xr_usb_serial_common.c:1867:17: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1867 |  .write_room =  xr_usb_serial_tty_write_room,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Running on Ubuntu 20.04.3 LTS