xmm7360/xmm7360-pci #138: Lenovo T490s, linux 5.13.13 segfault in insmod

mithodin, 2021-09-10:
I'm trying to use this driver on my Thinkpad T490s, but I am always getting a segfault when trying to load the kernel module. Here's the console output:

```
[lucas@mk1] ~/tools/xmm7360-pci $ make && make load
make -C /lib/modules/5.13.13_1/build M=/home/lucas/tools/xmm7360-pci modules
make[1]: Verzeichnis „/usr/src/kernel-headers-5.13.13_1“ wird betreten
make[1]: Verzeichnis „/usr/src/kernel-headers-5.13.13_1“ wird verlassen
sudo /sbin/rmmod xmm7360
Passwort: 
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Fehler 1 (ignoriert)
sudo /sbin/insmod xmm7360.ko
make: *** [Makefile:19: load] Speicherzugriffsfehler
[lucas@mk1] ~/tools/xmm7360-pci $ uname -a
Linux mk1 5.13.13_1 #1 SMP Fri Aug 27 13:28:13 UTC 2021 x86_64 GNU/Linux
```

Additionally, I see a kernel bug message in dmesg: 
[dmesg.log](https://github.com/xmm7360/xmm7360-pci/files/7142210/dmesg.log)

Please let me know if you need any additional info.

frcl, 2021-09-21:
I have the same error on my X13 Yoga with Kernel 5.14.6 and similar dmesg output.

tgxn, 2022-02-06:
This should be resolved now since the pull is merged, can this be tested @mithodin  @frcl  ?

mithodin, 2022-02-06:
@tgxn yes, I can now load the module. Not getting an ip via dhcp, though :thinking:

mithodin, 2022-02-06:
Update: had to input my pin again for some reason, it works now :-)