xmm7360/xmm7360-pci #141: compilation error on 5.14 kernel Debian

deelerke, 2021-09-22:
Hi all, I'm having this error on my freshly updated debian system now running 5.14 

root@bewardeb:/home/deeler/xmm7360-pci# make && make load
make -C /lib/modules/5.14.0-1-amd64/build M=/home/deeler/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-1-amd64'
  CC [M]  /home/deeler/xmm7360-pci/xmm7360.o
/home/deeler/xmm7360-pci/xmm7360.c:1322:16: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |  .write_room = xmm7360_tty_write_room,
      |                ^~~~~~~~~~~~~~~~~~~~~~
/home/deeler/xmm7360-pci/xmm7360.c:1322:16: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.14.0-1-common/scripts/Makefile.build:276: /home/deeler/xmm7360-pci/xmm7360.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.14.0-1-common/Makefile:1882: /home/deeler/xmm7360-pci] Error 2
make[1]: *** [/usr/src/linux-headers-5.14.0-1-common/Makefile:232: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-1-amd64'
make: *** [Makefile:9: default] Error 2
root@bewardeb:/home/deeler/xmm7360-pci#

deelerke, 2021-09-25:
I have found a solution/workaround myself:

add "-Wno-incompatible-pointer-types" to the Makefile
like this

`ccflags-y := -Wno-multichar -Wno-incompatible-pointer-types`

can this be rather a gcc10 error ?

teccem, 2021-09-25:
hi. 
https://github.com/xmm7360/xmm7360-pci/issues/137
 the same closed issue on Manjaro

my2018, 2021-10-10:
the same closed issue on Arch……

tgxn, 2022-02-10:
Hey @deelerke @my2018 

Now that these fixes have been merged, Can you please test that the master branch builds and works well with your current kernel versions?

naathnsmith, 2024-06-04:
Hey, I;m having a similar issue with an EPSOLAR Charge controller, when compiling the driver I get the following error.

`error: initialization of ‘ssize_t (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘long int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct tty_struct *, const unsigned char *, int)’ [-Werror=incompatible-pointer-types]
 1711 |         .write =                `xr_usb_serial_tty_write,`

I've tried making it so that the write_room function is unsigned, and cant find the edit for making the driver write actually work. How would I solve this?