xmm7360/xmm7360-pci #143: Substantial slow down and craches

mmohaveri, 2021-09-22:
When I load the kernel module, my laptop will become substantially slow, animations become glitchy, and browsers sometimes crash.

Meanwhile my CPU and Ram usage do not go up at all!

tgxn, 2022-02-06:
Hey @mmohaveri - And this is only happening while you're using this driver? What device/software are you running?

mmohaveri, 2022-02-13:
Yes, as soon as I load the kernel module my system slows down, and after a reset (which I believe will unload the module) system returns to normal speed.

I'm running a X1 Yoga 4th gen, with Debian unstable.

As I mentioned, the mystery is that my RAM usage does not go up (neither my CPU usage) only the system becomes unworkable and almost unresponsive. So I don't think it's a memory leak issue.