xmm7360/xmm7360-pci #148: OSError: FCC unlock failed

IronDusts, 2021-11-08:
Module compiled and loaded normally, trying to bring the interface up results into OSError: FCC unlock failed error. 

Device: 

55:00.0 Wireless controller [0d40]: **Intel Corporation XMM7360** LTE Advanced Modem (rev 01)
        Subsystem: Dell Device 5820
        Flags: bus master, fast devsel, latency 0, IRQ 170
        Memory at 8c400000 (64-bit, non-prefetchable) [size=4K]
        Memory at 8c401000 (64-bit, non-prefetchable) [size=1K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/4 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Latency Tolerance Reporting
        Capabilities: [150] L1 PM Substates
        Kernel driver in use: xmm7360

OS: 

[root@localhost ~]# uname -r 
**5.14.16-301.fc35.x86_64**
[root@localhost xmm7360-pci]# cat /etc/fedora-release 
**Fedora release 35 (Thirty Five)**
[root@localhost xmm7360-pci]# 

Laptop: 
[root@localhost xmm7360-pci]# dmidecode |grep -i latitud
	**Product Name: Latitude 5320**
	Family: Latitude

[root@localhost ~]# lte up
lte.sh: manage xmm7360-pci
**APN: internet.a1.bg**
Script: /home/k_georgiev/modem/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x19fb2faa
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c0204000001921100010102041100010102042e5829dd'
async_ack: 0x0
response: 0x0
Traceback (most recent call last):
  File "/home/k_georgiev/modem/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 53, in <module>
    rpc.do_fcc_unlock(r)
  File "/home/k_georgiev/modem/xmm7360-pci/rpc/rpc.py", line 388, in do_fcc_unlock
    raise IOError("FCC unlock failed")
**OSError: FCC unlock failed**

tgxn, 2022-02-06:
Does it work if you try bring the connection up `lte up` multiple times?
Also, are you able to attempt installing/booting to windows and re-connecting with the modem?

FCC lock is something that's toggled in the modem, you can see `FCC lock: state 0 mode 2`
There is some writing about it [here in the usb repo](https://github.com/xmm7360/xmm7360-usb-modeswitch#fcc-lock) specific to the FCC locking on this modem. I'm not sure if you're in a position to make use of it, but there's also [code there ](https://github.com/xmm7360/xmm7360-usb-modeswitch/blob/master/fcc_unlock/fcc_unlock.c) that I believe is meant to alter the FCC lock status.

tgxn, 2022-02-10:
Hey @IronDusts

I'm not really sure how FCC Locking works personally, as I've not had any device that's been locked.

In the `rpc.py` file of this repo, there is a method that is meant to perform the FCC Unlock command: https://github.com/xmm7360/xmm7360-pci/blob/master/rpc/rpc.py#L368 Though it appears this is where the code is failing in your case.

By the looks, it uses a key `key = bytearray([0x3d, 0xf8, 0xc7, 0x19])` to perform the unlocking, so I'm not sure why that wouldn't work for you.

I would say that testing under windows again would be ideal, and may remove the fcc_lock in the process. Otherwise I'm not too sure what we could try next.

IronDusts, 2022-02-17:
Switched from BIOS to USB mode, attached to KVM Windows 10 VM with USB passthrough. 
Modem was OK, drivers installed, still got message "No service". 
Removed the SIM PIN. 

Tried several times "Connect" and it didn't work, message was "No service" but it still knew somehow the mobile provider and APN. This makes me think it has service but just for some reason traffic was not passing. 

Didn't see anything else related to lock/unlock.

Rebooted, switched back to PCI mode in BIOS and tried again lte.sh setup && lte.sh up. 
Got the same message: 

lte.sh: manage xmm7360-pci
APN: internet.a1.bg
Script: /home/k_georgiev/modem/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x6eb9fac4
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c0204000001921100010102041100010102040364a915'
async_ack: 0x0
response: 0x0
Traceback (most recent call last):
  File "/home/k_georgiev/modem/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 53, in <module>
    rpc.do_fcc_unlock(r)
  File "/home/k_georgiev/modem/xmm7360-pci/rpc/rpc.py", line 388, in do_fcc_unlock
    raise IOError("FCC unlock failed")
OSError: FCC unlock failed

Not sure what should I do when I go to Windows.