xmm7360/xmm7360-pci #150: Kernel 5.15: Implicit declaration of function put_tty_driver

nursoda, 2021-11-16:
Since Arch switched kernel from 5.14 to 5.15, I can no longer compile the kernel module. This is what I see:

```
$ git pull
Already current.
$ make
make -C /lib/modules/5.15.2-arch1-1/build M=/BASE/xmm7360-pci modules
make[1]: Enter directory „/usr/lib/modules/5.15.2-arch1-1/build“
  CC [M]  /BASE/xmm7360-pci/xmm7360.o
/BASE/xmm7360-pci/xmm7360.c: In function »xmm7360_exit«:
/BASE/xmm7360-pci/xmm7360.c:1558:9: Error: Implicit declaration
of function »put_tty_driver« [-Werror=implicit-function-declaration]
 1558 |         put_tty_driver(xmm7360_tty_driver);
      |         ^~~~~~~~~~~~~~
cc1: Some warnings are handled as errors
make[2]: *** [scripts/Makefile.build:277: /BASE/xmm7360-pci/xmm7360.o] error 1
make[1]: *** [Makefile:1868: /BASE/xmm7360-pci/xmm7360-pci] error 2
make[1]: Exit directory „/usr/lib/modules/5.15.2-arch1-1/build“
make: *** [Makefile:9: default] error 2
```

Any hints how to *properly* solve this?

zear, 2021-11-20:
With kernel v5.15, `put_tty_driver` has been replaced with `tty_driver_kref_put`. Please build with this patch: https://github.com/xmm7360/xmm7360-pci/pull/152

> Any hints how to _properly_ solve this?

The _proper_ way would be to upstream this driver, so it doesn't have to be patched every time kernel API changes.

nursoda, 2021-11-20:
I confirm that fixes the issue on 5.15+ kernels. As long as this is a separate driver, should the code consider which kernel it is compiled for? And sure enough, drivers such as this should be shipped with the kernel. However I suspect its far away from code quality that would be accepted by kernel driver maintainers since it sometimes needs to be called, or even removed and re-added, and since there's way too much chatter output that would go into logs. Speaking for myself, I'm just an ordinary laptop user and do not know how to fixup the code properly. I was able to patch it though, and am very happy to be able to use my internal SIM card again. So, thanks a lot for investigating and the patch.

zear, 2021-11-20:
In case of this issue, `put_tty_driver` was an alias to an already existing `tty_driver_kref_put`, so replacing it with a call to the latter will not break compatibility with older kernels. However, that sort of issues will inevitably accumulate as the upstream kernel diverges from APIs used when this driver has been originally written. So the maintainers of this repository will have to decide whether they want to keep multiple branches for each API breaking kernel revision, or only support the mainline kernel.

Code cleanup in order to get this driver accepted upstream would be a great addition and I am willing to help there if the maintainers are interested. However, I think the bigger problem will be to get consent from every person who contributed to this project (the `Signed-off-by` line), which is mandatory for code to be accepted into the Linux kernel. Also potential issues with the origin of this driver (it being reverse engineered from a proprietary driver). But this goes beyond discussion of this issue. Glad to hear it solved your problem.

Jiivee1, 2021-11-25:
My laptop has Fibocom L850-GL,  which works with this xmm7360-pci driver. On their website, https://www.fibocom.com/en/products/LTECat-L850-GL.html, they tell that there is Linux driver (both Linux and Chrome OS)  for it. I can't find it from their website. I also tried to ask for the driver by email, but didn't get any response.

tgxn, 2022-02-10:
Howdy @nursoda

Can you please test that the `master` branch builds and works well with your current kernel versions?

nursoda, 2022-02-10:
Already did test: I pulled&merged the commits in my cloned repo in GitHub, then pulled my repo on my laptop, rebuilt and tested. Worked fine. My repo is a commit ahead though (open PR in yours).