xmm7360/xmm7360-pci #153: vmlinux isn't available

WGeffers, 2021-11-22:
I have a L14 Gen1 with a Ryzen 4750u.

When ever i try to make, make retruns:

"Skipping BTF generation for /home/.../tmp/xmm7360-pci/xmm7360.ko due to unavailability of vmlinux"

Tried on:
Ubuntu 20.04.3 and 21.10 and Fedora 35.

Maybe it's my fault, so if any one can help, please.

RealTehreal, 2022-01-31:
Just ignore the message. I'm on 20.04, get the same message, but it works anyway.

tgxn, 2022-02-06:
@WGeffers Are you having issues in getting a connection? What happens when you try to bring the connection up?

anilkvp, 2022-03-19:
I am also facing the issue. It was working fine for me earlier. anyway I am posting the make log below if its helpful

xmm7360-pci]$ make && make load
make -C /lib/modules/5.16.14-200.fc35.x86_64/build M=./xmm7360-pci modules
make[1]: Entering directory '/usr/src/kernels/5.16.14-200.fc35.x86_64'
  CC [M]  ./xmm7360-pci/xmm7360.o
  MODPOST ./xmm7360-pci/Module.symvers
  CC [M]  ./xmm7360-pci/xmm7360.mod.o
  LD [M]  ./xmm7360-pci/xmm7360.ko
  BTF [M] ./xmm7360-pci/xmm7360.ko
Skipping BTF generation for ./xmm7360-pci/xmm7360.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/5.16.14-200.fc35.x86_64'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko