xmm7360/xmm7360-pci #156: wwan0 created but not confiigured (possibly: no connection to provider)

pswiatki, 2021-12-12:
When I run `let up` script I get:
```
lte.sh: manage xmm7360-pci
APN: internet
Script: /home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'[REDACTED]'
response: 0x0
RPC executing UtaMsCbsInit
b'xxxxxxxxxxxx'
response: 0x0
RPC executing UtaMsNetOpen
b'xxxxxxxxxx'
response: 0x0
RPC executing UtaMsCallCsInit
b'xxxxxxxxxxxx'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'xxxxxxxxxxxxxxxx'
response: 0x0
RPC executing UtaMsSsInit
b'[REDACTED]'
response: 0x0
RPC executing UtaMsSimOpenReq
b'[REDACTED]'
response: 0x0
RPC executing CsiFccLockQueryReq
b'[REDACTED]'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'[REDACTED]'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b' ..... larger block of HEX values here
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'[REDACTED]'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
```
Later on, at somewhat regular intervals, these pop up one line at a time:
```
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x15, 0x1, 0x15, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x16, 0x1, 0x16, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x17, 0x1, 0x17, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x18, 0x1, 0x18, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x19, 0x1, 0x19, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x1a, 0x1, 0x1a, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x1b, 0x1, 0x1b, 0x1, 0x0, 0x0
```
My provider states a single word "internet" shall be used as APN name and user / password pair shall also be provided. 
What am I doing wrong to get no-comm (although the interface itself seems to be present in the system now)?

nursoda, 2022-01-14:
O₂? I also was told to use "internet", but it only works for me using an empty APN string ("").

pswiatki, 2022-01-14:
No, not O₂. It is ORANGE [Polska]. Anyhow, I will try without "internet" for APN and post the result.

pswiatki, 2022-01-15:
Unfortunately, the connection fails. Here is the output of `sudo ./lte.sh up`:

'''
[pswiatki@sdr-aptv scripts]$ sudo ./lte.sh up
lte.sh: manage xmm7360-pci
APN: 
Script: /home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x2, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
'''
big blob of hex data follows
then after the blob:
'''
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x4b, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3932058, 0x0, 0x3, 0x104, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0xffffffff, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x39, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3932058, 0x0, 0x3, 0x104, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x3939b8b, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x39, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3939b8b, 0x0, 0x3, 0x104, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
unsolicited: UtaMsNetRegistrationInfoIndCb: 0x0, 0x0, 0x0, 0x3, 0x104, 0x0, 0x22, 0x104, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x0, 0xffff, 0xffffffff, 0xffff, 0xffff, 0x0, 0x0, 0xfe, 0xfffe, 0x0, 0xffff, 0xffff, 0xfffe, 0x0, 0xffff, 0xffff, 0x0, 0x0, 0x39, 0x2, 0x0, 0x3, 0x104, 0x0, 0x0, 0xff, 0x1, 0x2, 0xff, 0x1, 0x2, 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0xff, 0x1, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x39, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3939b8b, 0x0, 0x3, 0x104, 0x0
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x4b, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3939b8b, 0x0, 0x3, 0x104, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0xffffffff, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x39, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3939b8b, 0x0, 0x3, 0x104, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x3939b8b, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x39, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0xd1cf, 0x3939b8b, 0x0, 0x3, 0x104, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
'''

Does it mean I am out of luck, or what?

nursoda, 2022-01-17:
No idea, sorry. I'm just a user as you are. I'm stuck too, not getting a connection since approx. kernel 5.14 on Arch.

pswiatki, 2022-01-17:
Understood, sure. I am just hoping the authors could chip in and explain what is going on here.
I am on ARCH, too. But this is the first time I tried this driver - just changed my laptop for a new Precision and I saw a picture of the WAN card in the service manual. So, I got it, installed it and then figured out that Network Manager does not see this interface automagically. It was my starting point with this interface in a Linux based system and how I arrived here.
Will keep trying - maybe there is actually something I could change in the setup to make it behave.

spikespaz, 2022-01-28:
I haven't any solution. Really hope this driver isn't abandoned.

tgxn, 2022-02-06:
Howdy! It's not in as active development as it used to be, mostly because it;s fairly stable, though I'm just accepting some pulls that could hopefully solve some of these new issues.

With regards to username/password authentication, this is a known issue https://github.com/xmm7360/xmm7360-pci/issues/62

pswiatki, 2022-02-23:
Rebased to cf6625ace8da1bd44f46eb71b2557ac01ee1d11a (current origin/master) and recompiled. Also, removed the PIN from the SIM. Now this is the output:

```
lte.sh: manage xmm7360-pci
APN: internet
Script: /home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x3, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000 [... cut ...]
name:xmm7360 uuid:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx type:generic
found Interface: wwan0
Managed: 0
activate
```
DNS'es proposed by the provider, but not taken into `/etc/resolv.conf`. I guess it is because Network Manager is in used.
Yes, I did enable:

```
# used to activate NetworkManager integration
dbus=True
```
How can I check which DNS'es NM actually uses?

Anyhow, it seems - for whatever reason - the PIN is not handled the way it should. I will take a closer look at it as the next step.

ordicker, 2022-03-11:
I have the same problem with only data sim, but with my phone sim it’s work perfectly.

Does someone have an idea what could be the problem?

ordicker, 2022-03-27:
#171 - I posted some logs in a new issue. Maybe that would be helpful

teccem, 2022-04-01:
Hello, a half of year ago i sed this driver and it was working, but last week i have tried install Linux one more time near Windows (i had to use Windows). I am using both OS and i got some issues with this driver too.
I also make post in [#171](https://github.com/xmm7360/xmm7360-pci/issues/171), but dublicate here.

Firmware: 18500.5001.00.05.27.12
may thmx changed in firmware?