xmm7360/xmm7360-pci #157: HP Elite x2 G4 Tablet support in Debian?

carrotcypher, 2021-12-16:
Following the instructions in this repo I get:

```
user@debian:~/Downloads/xmm7360-pci$ make && make load
make -C /lib/modules/5.15.0-2-amd64/build M=/home/user/Downloads/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-2-amd64'
  CC [M]  /home/user/Downloads/xmm7360-pci/xmm7360.o
/home/user/Downloads/xmm7360-pci/xmm7360.c:1322:23: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |         .write_room = xmm7360_tty_write_room,
      |                       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/Downloads/xmm7360-pci/xmm7360.c:1322:23: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
/home/user/Downloads/xmm7360-pci/xmm7360.c: In function ‘xmm7360_exit’:
/home/user/Downloads/xmm7360-pci/xmm7360.c:1558:9: error: implicit declaration of function ‘put_tty_driver’ [-Werror=implicit-function-declaration]
 1558 |         put_tty_driver(xmm7360_tty_driver);
      |         ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.15.0-2-common/scripts/Makefile.build:282: /home/user/Downloads/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-2-common/Makefile:1892: /home/user/Downloads/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-2-amd64'
make: *** [Makefile:9: default] Error 2
user@debian:~/Downloads/xmm7360-pci$ 

```

I've never had to work with linux headers before. How do I make sure I'm using the right one? Is that even the issue here?

nursoda, 2022-01-14:
See this closed bug: https://github.com/xmm7360/xmm7360-pci/issues/137

tgxn, 2022-02-10:
Hey @carrotcypher 

Now that these fixes have been merged, Can you please test that the `master` branch builds and works well with your current kernel versions?

carrotcypher, 2022-02-11:
@tgxn Still doesn't show any WWAN interface, but building seems to works fine.

`user@debian:~/Downloads/xmm7360-pci$ make && make load
make -C /lib/modules/5.15.0-2-amd64/build M=/home/user/Downloads/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-2-amd64'
  CC [M]  /home/user/Downloads/xmm7360-pci/xmm7360.o
  MODPOST /home/user/Downloads/xmm7360-pci/Module.symvers
  CC [M]  /home/user/Downloads/xmm7360-pci/xmm7360.mod.o
  LD [M]  /home/user/Downloads/xmm7360-pci/xmm7360.ko
  BTF [M] /home/user/Downloads/xmm7360-pci/xmm7360.ko
Skipping BTF generation for /home/user/Downloads/xmm7360-pci/xmm7360.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-2-amd64'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
user@debian:~/Downloads/xmm7360-pci$ sudo python3 rpc/open_xdatachannel.py --apn ****.*****.com
Traceback (most recent call last):
  File "/home/user/Downloads/xmm7360-pci/rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/user/Downloads/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
user@debian:~/Downloads/xmm7360-pci$ 
`