xmm7360/xmm7360-pci #161: Driver does not compile

tazhate, 2022-01-10:
```
➜  xmm7360-pci git:(master) ✗ make
make -C /lib/modules/5.15.6-051506-generic/build M=/home/taz/trash/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.6-051506-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-12ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  CC [M]  /home/taz/trash/xmm7360-pci/xmm7360.o
/home/taz/trash/xmm7360-pci/xmm7360.c:1322:16: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |  .write_room = xmm7360_tty_write_room,
      |                ^~~~~~~~~~~~~~~~~~~~~~
/home/taz/trash/xmm7360-pci/xmm7360.c:1322:16: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
/home/taz/trash/xmm7360-pci/xmm7360.c: In function ‘xmm7360_exit’:
/home/taz/trash/xmm7360-pci/xmm7360.c:1558:2: error: implicit declaration of function ‘put_tty_driver’ [-Werror=implicit-function-declaration]
 1558 |  put_tty_driver(xmm7360_tty_driver);
      |  ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/taz/trash/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1874: /home/taz/trash/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.6-051506-generic'
make: *** [Makefile:9: default] Error 2

➜  xmm7360-pci git:(master) ✗ uname -a
Linux tazthink 5.15.6-051506-generic #202112010437 SMP Wed Dec 1 09:47:58 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

➜  xmm7360-pci git:(master) ✗ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

```

MikeLindner, 2022-01-16:
~/t/xmm7360-pci (master |  🔍  2) 🐠  make                                                                                   10:38:03
make -C /lib/modules/5.15.15-1-lts/build M=/home/michael/temp/xmm7360-pci modules
make[1]: Entering directory '/usr/lib/modules/5.15.15-1-lts/build'
  CC [M]  /home/michael/temp/xmm7360-pci/xmm7360.o
/home/michael/temp/xmm7360-pci/xmm7360.c:1322:23: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |         .write_room = xmm7360_tty_write_room,
      |                       ^~~~~~~~~~~~~~~~~~~~~~
/home/michael/temp/xmm7360-pci/xmm7360.c:1322:23: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
/home/michael/temp/xmm7360-pci/xmm7360.c: In function ‘xmm7360_exit’:
/home/michael/temp/xmm7360-pci/xmm7360.c:1558:9: error: implicit declaration of function ‘put_tty_driver’ [-Werror=implicit-function-declaration]
 1558 |         put_tty_driver(xmm7360_tty_driver);
      |         ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/michael/temp/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1868: /home/michael/temp/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.15-1-lts/build'
make: *** [Makefile:9: default] Error 2

also does not compile on arch 5.15.15-1-lts
Compiles and loads on arch 5.10

ko-pp, 2022-01-18:
Hi, try the fix from #159 :)

tgxn, 2022-02-10:
Hey @tazhate @MikeLindner 

Now that these fixes have been merged, Can you please test that the master branch builds and works well with your current kernel versions?

martiinezz, 2022-02-20:
Compiles and works OK on 5.17-rc1 (didn't before)