xmm7360/xmm7360-pci #162: MAC address?

Zeltinger, 2022-01-18:
Hi!

Is is possible for the driver to set a MAC address?

thx
zeltinger

RealTehreal, 2022-02-02:
Just out of interest, what are you trying to achieve?

tgxn, 2022-02-06:
As far as I know it's not possible to override the modem's default MAC. There is at least no option in the driver for this.

Zeltinger, 2022-02-14:
Thanks for your comments.

We are using a mechanism based on MAC addresses in virtualization to tie host interfaces to specific interfaces in a VM. Currently the driver does not expose a MAC address which break our logic. So I was thinking to generate one for the modem based on some unique system information like the serial and set it before our logic kicks in.

If the modem has a fixed MAC address and it is just a matter of exposing it through the driver, that would be even better.

zeltinger