xmm7360/xmm7360-pci #164: Fix build error on kernel 5.15.15

Mad-Mason, 2022-01-24:
xmm7360.c fails to build on this kernel version due to 

1) a change in tty_driver.h which removed put_tty_driver [see here](https://lore.kernel.org/lkml/20210723074317.32690-8-jslaby@suse.cz/)
2) trying to write int pointer to unsigned int pointer

Errors from make
```
make -C /lib/modules/5.15.15/build M=/tmp/Package_Building/xmm7360/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-5.15.15'
  CC [M]  /tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.o
/tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.c:1322:23: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1322 |         .write_room = xmm7360_tty_write_room,
      |                       ^~~~~~~~~~~~~~~~~~~~~~
/tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.c:1322:23: note: (near initialization for ‘xmm7360_tty_ops.write_room’)
/tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.c: In function ‘xmm7360_exit’:
/tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.c:1558:9: error: implicit declaration of function ‘put_tty_driver’ [-Werror=implicit-function-declaration]
 1558 |         put_tty_driver(xmm7360_tty_driver);
      |         ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /tmp/Package_Building/xmm7360/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1868: /tmp/Package_Building/xmm7360/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/src/linux-5.15.15'
make: *** [Makefile:11: default] Error 2
```
Fixed by
1) change `put_tty_driver` to `tty_driver_kref_put` as the former was only an alias to the latter.
2) change `xmm7360_tty_write_room` to `unsigned int`. This seems OK as xmm7360_tty_write_room appears to only be used once so shouldn't break anything else.

Patch

```
--- xmm7360.c   2022-01-24 05:28:57.401524169 +0000
+++ xmm7360.c.new       2022-01-24 05:21:36.198178379 +0000
@@ -1272,7 +1272,7 @@
        return written;
 }
 
-static int xmm7360_tty_write_room(struct tty_struct *tty)
+static unsigned int xmm7360_tty_write_room(struct tty_struct *tty)
 {
        struct queue_pair *qp = tty->driver_data;
        if (!xmm7360_qp_can_write(qp))
@@ -1555,7 +1555,7 @@
        pci_unregister_driver(&xmm7360_driver);
        unregister_chrdev_region(xmm_base, 8);
        tty_unregister_driver(xmm7360_tty_driver);
-       put_tty_driver(xmm7360_tty_driver);
+       tty_driver_kref_put(xmm7360_tty_driver);
 }
 
 module_init(xmm7360_init);
```

Hope this helps.

RealTehreal, 2022-01-31:
Thank you very much - worked like a charm!

MaxG-Enpal, 2022-02-05:
Unfortunately, I get this error

mvg@max-ThinkPad-2-22:~/Downloads/xmm7360-pci$ make load
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Operation not permitted
make: *** [Makefile:19: load] Error 1

Mad-Mason, 2022-02-06:
I assume lsmod returns a list of your loaded modules I.e. your kernel supports modules.

I’d try (as root)

modinfo xmm7360

to make sure module is installed OK (should be in ‘lib/modules/`uname -r`/extras).

If that works then try

modprobe xmm7360

If it doesn’t, try

depmod -A
modprobe xmm7360
lsmod | grep `xmm7360`

MaxG-Enpal, 2022-02-06:
> I assume lsmod returns a list of your loaded modules I.e. your kernel supports modules.
>
> I’d try (as root)
>
> modinfo xmm7360
>
> to make sure module is installed OK (should be in ‘lib/modules/`uname [...]

I get : modinfo: ERROR: Module xmm7360 not found.

Mad-Mason, 2022-02-07:
Can you find the module under ‘/lib/modules/5.15.0-10-generic (should be in the ‘extras’ subdirectory)?
If not, the ‘make’ failed to build and install this module. To try and diagnose the problem you would need to run the ‘make’ again and post the output.

tgxn, 2022-02-10:
Hey @Mad-Mason  @MaxG-Enpal  @RealTehreal 

Now that these fixes have been merged, Can you please test that the `master` branch builds and works well with your current kernel versions?

RealTehreal, 2022-02-10:
> Hey @Mad-Mason @MaxG-Enpal @RealTehreal
> 
> Now that these fixes have been merged, Can you please test that the `master` branch builds and works well with your current kernel versions?

For me, it works again. Thank you for maintaining this driver :+1:

Mad-Mason, 2022-02-10:
I can confirm that it works.

Many thanks
Iain