xmm7360/xmm7360-pci #171: ERROR:root:Attach failed again, giving up ;No PIN and logs

ordicker, 2022-03-21:
I got two SIMs and one of them is working and the other doesn't.(the one that doesn't is Data only)
The one that doesn't work has two problems:
1. When running `lte up` for the first time, I get: 
....
`
unsolicited: UtaMsNetConnectionInfoIndCb: 0x0, 0x2, 0x3, 0x0, 0x0, 0x1, 0x0, 0x0
unsolicited: UtaMsNetConnectionInfoIndCb: 0x0, 0x2, 0x3, 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x1, 0x35d, 0x0, 0x67, 0x3, 0x0, 0x0, 0xff, 0x0, 0x0, 0x0, 0x0, 0x0, 0xffff, 0x3a3520, 0x0, 0x9, 0x1a9, 0x0
unsolicited: UtaMsNetConnectionInfoIndCb: 0x0, 0x2, 0x3, 0x0, 0x0, 0x1, 0x0, 0x0
unsolicited: UtaMsNetConnectionInfoIndCb: 0x0, 0x2, 0x3, 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x1, 0x35d, 0x0, 0x67, 0x3, 0x0, 0x0, 0xff, 0x0, 0x0, 0x0, 0x0, 0x0, 0xffff, 0x3a3520, 0x0, 0x9, 0x1a9, 0x0
`
...
For ever. 
2. When I run it for the second time (after Ctrl-C) I get: 
`
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x1, 0x35d, 0x0, 0x67, 0x3, 0x0, 0x0, 0xff, 0x0, 0x0, 0x0, 0x0, 0x0, 0xffff, 0x3a3520, 0x0, 0x9, 0x1a9, 0x0
response: 0x0, 0xffffffff, 0x0, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
`
Things that I tried:
1. switching the SIMs (the working one is always working) 
2. Using the non working SIM on my phone. Works. 
3. No PIN lock (checked many times) 

Computer: Lenovo ThinkPad X1 Carbon 7th
OS: Ubuntu 20.04

working log: 
`
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: we
Script: /home/dicker/tmp/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0xbcc119b8
[... 117 lines cut ...]
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x4b, 0x2, 0x0, 0x2, 0x5, 0x1, 0x1, 0x0, 0x0, 0x0, 0x1839, 0x302d87, 0x0, 0x3, 0x1a9, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0xffffffff, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x36, 0x2, 0x0, 0x2, 0x5, 0x1, 0x1, 0x0, 0x0, 0x0, 0x1839, 0x302d87, 0x0, 0x3, 0x1a9, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x302d80, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x36, 0x2, 0x0, 0x2, 0x5, 0x1, 0x1, 0x0, 0x0, 0x0, 0x1839, 0x302d80, 0x0, 0x3, 0x1a9, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
unsolicited: UtaMsNetRegistrationInfoIndCb: 0x0, 0x0, 0x0, 0x9, 0x1a9, 0x0, 0x2, 0x1a9, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x1, 0x38f, 0x0, 0x9, 0x1a9, 0x0, 0x0, 0xff, 0xffff, 0xffffffff, 0xffff, 0xffff, 0x8477, 0x0, 0x2, 0x1a9, 0x0, 0x0, 0x36, 0x2, 0x0, 0x3, 0x1a9, 0x0, 0x0, 0xff, 0x1, 0x2, 0xff, 0x1, 0x2, 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0xff, 0x1, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x36, 0x2, 0x0, 0x2, 0x5, 0x1, 0x1, 0x0, 0x0, 0x0, 0x1839, 0x302d80, 0x0, 0x3, 0x1a9, 0x0
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x36, 0x2, 0x0, 0x2, 0x5, 0x1, 0x1, 0x0, 0x0, 0x0, 0x1839, 0x302d80, 0x0, 0x3, 0x1a9, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
`
I saw some issues that could be related to this problem. Here I just gathered all the information I could find.
BTW, great work and thanks for the driver!

teccem, 2022-04-01:
Hello, a half of year ago i sed this driver and it was working, but last week i have tried install Linux one more time near  Windows  (i had to use Windows). I am using both OS and i got some issues with this driver too.
1) First run programm hungs on
 RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
CRTL+C
2) second  times it  hang on UtaMsSimInitIndCb (i don't remember exactly and redact in near future). I stop programm with CTRL+C and got the same issue as **ordicker** on 3th run and next more

ordicker, 2022-04-01:
Are the three SIMs -data only?
Because when I use my phone’s SIM, it does work.

teccem, 2022-04-01:
1 - data only, other 2 - full-operational

kunamashina, 2022-04-03:
Hi and thanks for the enormous effort to give us this driver.

I have the same issue running kernel version 5.17.1-1-default on OpenSUSE Tumbleweed and I have been troubleshooting for 3 days now without success

My sim has no PIN - using sim from O2 in Germany 

[log-lte.txt](https://github.com/xmm7360/xmm7360-pci/files/8405828/log-lte.txt)

ordicker, 2022-04-04:
I did another experiment.
I took the SIM that didn't work and changed this plan (from data only to regular plan), and it is just works!

Does someone have a clue why that could happen?

teccem, 2022-04-07:
i had the mistake in my first comment:
"INFO:root:Attach failed - waiting to see if we just weren't ready" - rising any time. 
"unsolicited: UtaMsSmsSimMsgCacheFinishedIndCb" - appeared do to logging latency after.

connection log output example:

---- to long byte code ----
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready

in this code part:
_, status = rpc.unpack('nn', attach['body'])

if status == 0xffffffff:
    logging.info("Attach failed - waiting to see if we just weren't ready")

    while not r.attach_allowed:
        r.pump()

i have try to set "attach_allowed" to true, but reasonably it rise "Attach failed again, giving up".  Other my attempts to debug didn't bring any usefull also.

The provider in manual say to set user, password, and authentication type. But as i say in my first comment the driver works with the same config before with all theses  SIM's.

teccem, 2022-04-30:
I don't know what exactly happen but:

    i had decision - smth with pytnon3. Last year all work, but now - not. I tied to install pthon3.8 instead of 3.9 or 3.10
    i had install ONLY pyroute2 and configparse from REPO, not PIP.

AND NOW IN WORK!

teccem, 2022-05-05:
Hello. I am sorry for last post, i have mistaken.
I really don't know reason but:
 if on Windows Cell network is disabled - this python script can't connect to device and write "INFO:root:Attach failed - waiting to see if we just weren't ready". I spend a week to dig it, 
 To maked it work i installed Windows, enabled Cell network and reinstalled Linux again. I didn't  find another way

Einstonoy, 2023-08-12:
> Hello. I am sorry for last post, i have mistaken.
> I really don't know reason but:
> if on Windows Cell network is disabled - this python script can't connect to device and write "INFO:root:Attach [...]
yeah that is perfectly work now but reboot unwork.