xmm7360/xmm7360-pci #172: Kernel panic when I take too much time for disk decryption

joushx, 2022-04-03:
When it takes me longer than a few seconds to enter the LUKS encryption password at boot (e.g. when needing multiple attempts), a kernel panic occurs afterwards at `xmm7360_irq0+0x7f`:

[image]

This is the output of the dump:

```
0000000000001930 <xmm7360_irq0>:
{
    1930:	e8 00 00 00 00       	call   1935 <xmm7360_irq0+0x5>
    1935:	41 57                	push   %r15
    1937:	41 56                	push   %r14
    1939:	41 55                	push   %r13
    193b:	41 54                	push   %r12
    193d:	49 89 f4             	mov    %rsi,%r12
    1940:	55                   	push   %rbp
    1941:	53                   	push   %rbx
    1942:	48 83 ec 18          	sub    $0x18,%rsp
	if (xmm->cp->status.code == 0xbadc0ded) {
    1946:	48 8b 46 60          	mov    0x60(%rsi),%rax
    194a:	8b 40 38             	mov    0x38(%rax),%eax
    194d:	3d ed 0d dc ba       	cmp    $0xbadc0ded,%eax
    1952:	0f 84 00 00 00 00    	je     1958 <xmm7360_irq0+0x28>
	if (xmm->bar2[BAR2_STATUS] != 0x600df00d) {
    1958:	49 8b 44 24 18       	mov    0x18(%r12),%rax
    195d:	8b 10                	mov    (%rax),%edx
    195f:	81 fa 0d f0 0d 60    	cmp    $0x600df00d,%edx
    1965:	0f 85 00 00 00 00    	jne    196b <xmm7360_irq0+0x3b>
	wake_up(&xmm->wq);
    196b:	31 c9                	xor    %ecx,%ecx
    196d:	49 8d 7c 24 28       	lea    0x28(%r12),%rdi
    1972:	ba 01 00 00 00       	mov    $0x1,%edx
    1977:	be 03 00 00 00       	mov    $0x3,%esi
    197c:	e8 00 00 00 00       	call   1981 <xmm7360_irq0+0x51>
	if (xmm->td_ring) {
    1981:	49 83 fc 90          	cmp    $0xffffffffffffff90,%r12
    1985:	0f 84 ad 03 00 00    	je     1d38 <xmm7360_irq0+0x408>
	if (!xmm->net)
    198b:	49 8b 84 24 70 2b 02 	mov    0x22b70(%r12),%rax
    1992:	00 
    1993:	48 85 c0             	test   %rax,%rax
    1996:	0f 84 f3 01 00 00    	je     1b8f <xmm7360_irq0+0x25f>
	qp = xmm->net->qp;
    199c:	48 8b 78 08          	mov    0x8(%rax),%rdi
    19a0:	49 8b 84 24 78 2b 02 	mov    0x22b78(%r12),%rax
    19a7:	00 
    19a8:	4c 8b 80 00 04 00 00 	mov    0x400(%rax),%r8
	ring = &xmm->td_ring[qp->num * 2 + 1];
    19af:	8b 97 d0 04 00 00    	mov    0x4d0(%rdi),%edx
	qp = xmm->net->qp;
    19b5:	48 89 7c 24 10       	mov    %rdi,0x10(%rsp)
	struct xmm_dev *xmm = qp->xmm;
```

The result address should then (as far as I understood the output) be `0x19AF` which is `ring = &xmm->td_ring[qp->num * 2 + 1];`

teccem, 2022-04-07:
joushx, check secure boot, if you are using it, signing xmm7360.ko. The second version try to use LTS kernel. I had a lot similar problems with not-LTS kernel and full disk encryption on Manjaro.

joushx, 2022-04-08:
Secure boot is disabled, but the LTS kernel solved the problem. Thanks!

closed 2022-04-08 by joushx