xmm7360/xmm7360-pci #173: INFO:root:Attach failed - waiting to see if we just weren't ready

anfty, 2022-04-07:
- Computer model: ThinkPad X390.  
- System: Linux X390 5.13.0-37-generic #42~20.04.1-Ubuntu.   

After running **lte up**, the driver does not work properly. I don't know what's going on. What do I do.

```
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready

```

teccem, 2022-04-17:
i got the same issue. I tried to use LTS kernels from 5.4 to 5.15. on Manjaro, Opensuse, Mandriva.

Log:

```
bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0xe, 0x1, 0x1, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x5, 0xf, 0x91, 0x1, 0x0, 0x1, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x3, 0x0, 0x90, 0x2, 0x1, 0x2, 0x1, 0x0, 0x0
unsolicited: UtaMsSimFullAccessIndCb: 0x0
unsolicited: UtaMsSmsSimMsgCacheFinishedIndCb: 0x0, 0x0
```

adaasch, 2022-04-29:
Any news or ideas on this? I'm also stuck at `INFO:root:Attach failed - waiting to see if we just weren't ready` since a couple of months...

teccem, 2022-04-30:
I don't know what exactly happen but:
1) i had decision - smth with pytnon3. Last year all work, but now - not. I tied to install pthon3.8 instead of 3.9 or 3.10
2) i had install ONLY pyroute2 and configparse from REPO, not PIP.

AND NOW IN WORK!

teccem, 2022-05-05:
Hello. I am sorry for last post, i have mistaken.
I really don't know reason but:
 if on Windows Cell network is disabled - this python script can't connect to device and write "INFO:root:Attach failed - waiting to see if we just weren't ready". I spend a week to dig it, 
 To maked it work i installed Windows, enabled Cell network and reinstalled Linux again. I didn't  find another way

teccem, 2022-05-23:
Yes, i have the same on first start. It like hang's up. I rewrite lte.sh script a little. It checks wwna0 interface and create/delete it accordingly. 

cat /usr/local/etc/xmm7360/scripts/lte-switch.sh 
#!/bin/bash
# install: sudo ./scripts/lte.sh setup
# use: sudo lte up

SCRIPT_PATH=$(readlink -f $0)
SCRIPT_DIR=`dirname $SCRIPT_PATH`
CONF_FILE=$SCRIPT_DIR/../xmm7360.ini

# check if xmm7360.ini is available or exit
if [ -f "$CONF_FILE" ]; then
  source $CONF_FILE
else
  echo "no configuration file found, you can create it by copying the sample file like this:"
  echo "cp $CONF_FILE.sample $CONF_FILE"
  exit 1
fi

ip link show wwan0

if [ $? == 1 ]
   then
      cd $SCRIPT_DIR/..
      make
      make load
      sleep 2
      python3 $SCRIPT_DIR/../rpc/open_xdatachannel.py
      ip link set wwan0 up
else
  ip link set wwan0 down
  rmmod xmm7360 || true
fi

Have added this script to systemd service, have allowed to my user run this service as sudo without password  and have created .desctop file.