xmm7360/xmm7360-pci #174: Works on Thinkpad L14 Gen 1 running fedora 35 kde spin

eniomarcosm, 2022-04-10:
It is running on Fedora 35 (and I think it should be running on other distros like Manjaro, Ubuntu and Arch Based). Although still need a interface to switch on, integrated with the DE. 
Thank you for making this project available, a big **GITHUG**.

Lyton505, 2022-04-10:
Are you by any chance using Econet as your mobile provider? I am having difficulties and I'm suspecting that it's probably my mobile provider. Asked the question here: https://askubuntu.com/questions/1401005/how-to-get-sim-connection-activated

eniomarcosm, 2022-04-18:
Did you remove the PIN?

spikespaz, 2022-04-25:
Could you please list any troubleshooting steps and/or special configuration that you had to do? The instructions on the repo aren't working for me.

perceival, 2022-05-16:
Works as well on HP EliteBook 745 G6 under Fedora 36. Here are the steps to set it up:
```
sudo dnf -y groupinstall "Development Tools" "Development Libraries"
sudo dnf -y downgrade gcc
git clone https://github.com/xmm7360/xmm7360-pci.git
cd xmm7360-pci
make
cp xmm7360.ini.sample xmm7360.ini
nano xmm7360.ini
/usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ~/certs/signing_key.pem ~/certs/signing_key.x509 xmm7360.ko # only needed with secure boot
sudo make load # redundant if setup script below works fine
./scripts/lte.sh setup
./scripts/lte.sh up
resolvectl dns wwan0 1.1.1.1 # or your preferred resolver
```
Above is neither compatible with ModemMenager nor NetworkManager. Connection status can be verified with `ip address`