xmm7360/xmm7360-pci #175: '/dev/xmm0/rpc' missing when using 'python3 rpc/open_xdatachannel.py'

annowak, 2022-04-14:
I am using DELL Latitiude 5410 with Ubuntu 18.04.2 LTS. Following manual gives me error at step:
`python3 rpc/open_xdatachannel.py`
Error massage:
```
andrzej@andrzej-Latitude-5410:~/tmp/xmm7360-pci$ sudo python3 rpc/open_xdatachannel.py
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/andrzej/tmp/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
```
I have searched the internet but have not found a solution. Perhaps I do not know what to look for.

anfty, 2022-04-16:
It looks like xMM7360 module does not exist, you should check xMM7360 before executing open_xdatachannel

annowak, 2022-04-20:
I looked on the manufacturer's website using my 'service tag' and indeed this module is missing.
`556-BBCD : No mobile broadband card`
I have the option to enter a SIM card which has misled me into thinking that a WWAN module has also been mounted.

Thank you for your help.

closed 2022-04-20 by annowak

ch4m30, 2022-10-11:
I have a Lenovo P53, with the Fibocom 850GL, and am receiving the exact same error as above. I have attempted to install the necessary packages via AUR (as well as compiling / building manually). 

Much the same as @annowak - have searched a lot of places, but am unfortunately coming up empty.