xmm7360/xmm7360-pci #176: /dev/wwan0 not being created

spikespaz, 2022-04-25:
When I run `lte up` I see the following output.
```
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: your.apn.here
Script: /home/jacob/Desktop/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
Traceback (most recent call last):
  File "/home/jacob/Desktop/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/jacob/Desktop/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
Cannot find device "wwan0"
```

I am running Manjaro KDE on 5.17. I don't know if the kernel is too recent. I have headers and DKMS installed.

pascalharp, 2022-05-19:
I get the exact same error on just arch linux on a lenovo x1 yoga (gen 3). I tried it on the lates kernel (5.17) as well as on the lts kernel (5.15). /dev/ttyXMM1 does not show up either.

iMonZ, 2022-05-26:
> I get the exact same error on just arch linux on a lenovo x1 yoga (gen 3). I tried it on the lates kernel (5.17) as well as on the lts kernel (5.15). /dev/ttyXMM1 does not show up either.

Tried compiling with 5.15 kernel.
Incompatible pointer type

KataFF, 2023-08-19:
This interface is there only when `iosm` driver is loaded. `/dev/ttyXMM*` devices get created when switching the modem to USB mode via: https://github.com/xmm7360/xmm7360-usb-modeswitch

Maybe the trick is to load `iosm` first and run the `lte.up` script so that it changes the module itself?