xmm7360/xmm7360-pci #178: Out of memory when transferring larger amounts of data

Jiivee1, 2022-05-27:
I have tested xmm7360 driver with larger data amount transfers.

My machine (Lenovo T14s) has 16 GiB RAM and 17 GiB swap. After transferring 8–9 GB from that machine to network, machine runs out of memory. It only happens if I am using xmm7360 driven LTE network. Kernel logs by themselves don't seem to log anything xmm7360 related, but only out of memory related stuff. I can reproduce this error.  I noticed that soon after swap usage exceeded 1.3 GB in otherwise silent than transferring data, machine ran out of RAM.

My cloned xmm7360 tree is uptodate.

Is there anything I can help with to get this memory leak solved?

```
$ uname -srvmpio
Linux 5.17.9-1-default #1 SMP PREEMPT Wed May 18 10:03:12 UTC 2022 (eab1a2c) x86_64 x86_64 x86_64 GNU/Linux
$ cat /etc/os-release 
NAME="openSUSE Tumbleweed"
# VERSION="20220525"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20220525"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20220525"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo-Tumbleweed"

```

Jiivee1, 2022-05-28:
It seems that transferring data from my machine to network massively leaks kernel memory. To reproduce:
1. Reboot machine and close unnecessary apps
2. Open LTE connection using xmm7360 driver
3. Transfer data to network. I tested like this:

`dd if=/dev/zero | ssh user@host.example.com "dd iflag=fullblock status=progress of=/dev/null"`

4. Launch top and monitor available memory vanishing
5. Interrupt dd before OOM (for example when available memory around is 0.5 GB)
6. Shut down LTE connection
7. Observe that occupied memory is not released
8. Reboot to get to fresh state.

ge0rg, 2022-10-19:
previous discussion and analysis of the memory leak in #131