xmm7360/xmm7360-pci #179: Stopped working (compilation failure) after Linux 5.18.0-arch1-1 update

0ln, 2022-05-28:
I installed the latest Arch Linux kernel a few days ago and noticed the modem wasn't starting up anymore.
After some troubleshooting, I figured out compilation now seems to fail.
I can't assure it is linked to the kernel update, but it happened right after it so I guess it's probable.
```bash
# sudo systemctl status xmm7360.service
× xmm7360.service - XMM7360 Modem Init
     Loaded: loaded (/usr/lib/systemd/system/xmm7360.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2022-05-28 10:01:02 CEST; 10min ago
    Process: 41718 ExecStart=/usr/lib/xmm7360-pci-git/rpc/open_xdatachannel.py -c /etc/xmm7360 (code=exited, status=1/FAILURE)
   Main PID: 41718 (code=exited, status=1/FAILURE)
        CPU: 199ms

May 28 10:01:01 carbon systemd[1]: Starting XMM7360 Modem Init...
May 28 10:01:02 carbon open_xdatachannel.py[41718]: Traceback (most recent call last):
May 28 10:01:02 carbon open_xdatachannel.py[41718]:   File "/usr/lib/xmm7360-pci-git/rpc/open_xdatachannel.py", line 42, in <module>
May 28 10:01:02 carbon open_xdatachannel.py[41718]:     r = rpc.XMMRPC()
May 28 10:01:02 carbon open_xdatachannel.py[41718]:   File "/usr/lib/xmm7360-pci-git/rpc/rpc.py", line 19, in __init__
May 28 10:01:02 carbon open_xdatachannel.py[41718]:     self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
May 28 10:01:02 carbon open_xdatachannel.py[41718]: FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
May 28 10:01:02 carbon systemd[1]: xmm7360.service: Main process exited, code=exited, status=1/FAILURE
May 28 10:01:02 carbon systemd[1]: xmm7360.service: Failed with result 'exit-code'.
May 28 10:01:02 carbon systemd[1]: Failed to start XMM7360 Modem Init.
```
```bash
# yay -S xmm7360-pci-dkms-git
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur:1]  xmm7360-pci-git-r229.cf6625a-1 (xmm7360-pci-dkms-git)

  1 xmm7360-pci-git (xmm7360-pci-dkms-git)   (Installed) (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
:: PKGBUILD up to date, Skipping (1/0): xmm7360-pci-git
  1 xmm7360-pci-git (xmm7360-pci-dkms-git)   (Installed) (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
:: (1/1) Parsing SRCINFO: xmm7360-pci-git (xmm7360-pci-dkms-git)
==> Making package: xmm7360-pci-git r229.cf6625a-1 (Sat 28 May 2022 10:04:05 CEST)
==> Retrieving sources...
  -> Updating xmm7360-pci git repo...
Fetching origin
  -> Found dkms.conf
  -> Found xmm7360.service
  -> Found nodbus-exit-code.patch
  -> Found dns-priority.patch
==> Validating source files with sha256sums...
    xmm7360-pci ... Skipped
    dkms.conf ... Passed
    xmm7360.service ... Passed
    nodbus-exit-code.patch ... Passed
    dns-priority.patch ... Passed
==> Making package: xmm7360-pci-git r229.cf6625a-1 (Sat 28 May 2022 10:04:07 CEST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating xmm7360-pci git repo...
Fetching origin
  -> Found dkms.conf
  -> Found xmm7360.service
  -> Found nodbus-exit-code.patch
  -> Found dns-priority.patch
==> Validating source files with sha256sums...
    xmm7360-pci ... Skipped
    dkms.conf ... Passed
    xmm7360.service ... Passed
    nodbus-exit-code.patch ... Passed
    dns-priority.patch ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of xmm7360-pci git repo...
Cloning into 'xmm7360-pci'...
done.
==> Starting prepare()...
patching file rpc/open_xdatachannel.py
patching file rpc/open_xdatachannel.py
==> Starting pkgver()...
==> Sources are ready.
==> Making package: xmm7360-pci-git r229.cf6625a-1 (Sat 28 May 2022 10:04:10 CEST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting pkgver()...
==> Sources are ready.
 -> xmm7360-pci-git-r229.cf6625a-1 already made -- skipping build
 -> Found git repo: github.com/xmm7360/xmm7360-pci
loading packages...
warning: xmm7360-pci-dkms-git-r229.cf6625a-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) xmm7360-pci-dkms-git-r229.cf6625a-1

Total Installed Size:  0.04 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                                                                  [###################################################################] 100%
(1/1) checking package integrity                                                                                [###################################################################] 100%
(1/1) loading package files                                                                                     [###################################################################] 100%
(1/1) checking for file conflicts                                                                               [###################################################################] 100%
(1/1) checking available disk space                                                                             [###################################################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove xmm7360-pci-git/r229.cf6625a
:: Processing package changes...
(1/1) reinstalling xmm7360-pci-dkms-git                                                                         [###################################################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install --no-depmod xmm7360-pci-git/r229.cf6625a -k 5.18.0-arch1-1
Error! Bad return status for module build on kernel: 5.18.0-arch1-1 (x86_64)
Consult /var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/make.log for more information.
==> WARNING: `dkms install --no-depmod xmm7360-pci-git/r229.cf6625a -k 5.18.0-arch1-1' exited 10
```
```bash
# cat /var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/make.log
DKMS make.log for xmm7360-pci-git-r229.cf6625a for kernel 5.18.0-arch1-1 (x86_64)
Sat 28 May 10:04:12 CEST 2022
make -C /usr/lib/modules/5.18.0-arch1-1/build M=/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build modules
make[1]: Entering directory '/usr/lib/modules/5.18.0-arch1-1/build'
  CC [M]  /var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.o
/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.c: In function ‘xmm7360_irq0’:
/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.c:1175:13: warning: the comparison will always evaluate as ‘true’ for the address of ‘td_ring’ will never be NULL [-Waddress]
 1175 |         if (xmm->td_ring) {
      |             ^~~
/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.c:209:24: note: ‘td_ring’ declared here
  209 |         struct td_ring td_ring[16];
      |                        ^~~~~~~
/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.c: In function ‘xmm7360_probe’:
/var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.c:1455:15: error: implicit declaration of function ‘pci_set_dma_mask’ [-Werror=implicit-function-declaration]
 1455 |         ret = pci_set_dma_mask(dev, 0xffffffffffffffff);
      |               ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build/xmm7360.o] Error 1
make[1]: *** [Makefile:1834: /var/lib/dkms/xmm7360-pci-git/r229.cf6625a/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.18.0-arch1-1/build'
make: *** [Makefile:9: default] Error 2
```
```bash
# uname -rs
Linux 5.18.0-arch1-1
```
Note: I'm on a ThinkPad X1 Carbon 7th gen.

JoelLinn, 2022-05-30:
Internal kernel APIs changed but for some reason it still does not create  /dev/xmm0

```diff
diff --git a/xmm7360.c b/xmm7360.c
index b23d660..f571d9c 100644
--- a/xmm7360.c
+++ b/xmm7360.c
@@ -1452,12 +1452,11 @@ static int xmm7360_probe(struct pci_dev *dev, const struct pci_device_id *id)
        }
        pci_set_master(dev);
 
-       ret = pci_set_dma_mask(dev, 0xffffffffffffffff);
+       ret = dma_set_mask_and_coherent(xmm->dev, DMA_BIT_MASK(64));
        if (ret) {
                dev_err(xmm->dev, "Cannot set DMA mask\n");
                goto fail;
        }
-       dma_set_coherent_mask(xmm->dev, 0xffffffffffffffff);
 
        ret = pci_request_region(dev, 0, "xmm0");
        if (ret) {

```

JoelLinn, 2022-05-31:
Thanks I will open a PR then.
Good catch about iosm

thotypous, 2022-06-16:
> Seems that a firmware release is anticipated for the L860, perhaps it's also needed for the L850 before it will work with ModemManager.. (?)

It surely isn't a firmware issue. `/dev/wwan0at0` and `/dev/wwan0at1` are there and work as expected (see below). The `wwan0` network interface is also there. Problem is that ModemManager does not seem to know it is supposed to manage `/dev/wwan0at*`: `mmcli -L` returns `No modems were found`. So it's a ModemManager issue.

```
sudo picocom --baud 115200 --echo /dev/wwan0at0
picocom v3.1

port is        : /dev/wwan0at0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : yes
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        :
omap is        :
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
ATI3

"Fibocom LTE Module"

OK
ATI5

"XMM7360"

OK
ATI7

"L850 LTE Module"

OK
ATI8

"18500.5001.00.01.20.75"

OK
```

thotypous, 2022-06-16:
@regenman

> Seems that a firmware release is anticipated for the L860, perhaps it's also needed for the L850 before it will work with ModemManager.. (?)

It surely isn't a firmware issue. `/dev/wwan0at0` and `/dev/wwan0at1` are there and work as expected (see below). The `wwan0` network interface is also there. Problem is that ModemManager does not seem to know it is supposed to manage `/dev/wwan0at*`: `mmcli -L` returns `No modems were found`. So it's a ModemManager issue.

```
$ sudo picocom --baud 115200 --echo /dev/wwan0at0
picocom v3.1

port is        : /dev/wwan0at0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : yes
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        :
omap is        :
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
ATI3

"Fibocom LTE Module"

OK
ATI5

"XMM7360"

OK
ATI7

"L850 LTE Module"

OK
ATI8

"18500.5001.00.01.20.75"

OK
```

closed 2022-08-12 by abrasive

yummy527, 2022-09-07:
According to #181 , I blocked iosm. And only xmm7360_usb is in use now.
```
# lspci -k
02:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
        Subsystem: Device 1cf8:8505
        Kernel driver in use: xmm7360_usb
        Kernel modules: iosm, xmm7360_usb
```
However, I also got the error message "FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'".
```shell
archcarbon% lte up
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: ctlte
Script: /home/xxxx/Downloads/xmm7360-pci-master/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
Traceback (most recent call last):
  File "/home/xxxx/Downloads/xmm7360-pci-master/scripts/../rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/xxxx/Downloads/xmm7360-pci-master/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
Cannot find device "wwan0"

```

Below is machine and OS info.
PC brand: Thinkpad X1 carbon gen8 with 4G modem XMM7360.
OS: arch linux (Linux 5.19.7-arch1-1)

Did I missed anything? What should I check?
Thanks.

nursoda, 2022-09-07:
Strange, it builds and works fine for me, also on 5.19.7-arch1-1, on a T14s Gen1 with the same modem:
`05:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)`

This is what I see during compilation:
```
make -C /lib/modules/5.19.7-arch1-1/build M=/PATH/xmm7360-pci clean
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird betreten
  CLEAN   /PATH/xmm7360-pci/Module.symvers
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird verlassen
make -C /lib/modules/5.19.7-arch1-1/build M=/PATH/xmm7360-pci modules
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird betreten
  CC [M]  /PATH/xmm7360-pci/xmm7360.o
/PATH/xmm7360-pci/xmm7360.c: In Funktion »xmm7360_irq0«:
/PATH/xmm7360-pci/xmm7360.c:1175:13: Warnung: der Vergleich wird immer als »true« ausgewertet, da die Adresse von »td_ring« niemals NULL sein wird [-Waddress]
 1175 |         if (xmm->td_ring) {
      |             ^~~
/PATH/xmm7360-pci/xmm7360.c:209:24: Anmerkung: »td_ring« ist hier deklariert
  209 |         struct td_ring td_ring[16];
      |                        ^~~~~~~
  MODPOST /PATH/xmm7360-pci/Module.symvers
  CC [M]  /PATH/xmm7360-pci/xmm7360.mod.o
  LD [M]  /PATH/xmm7360-pci/xmm7360.ko
  BTF [M] /PATH/xmm7360-pci/xmm7360.ko
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird verlassen
make -C /lib/modules/5.19.7-arch1-1/build M=/PATH/xmm7360-pci modules_install
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird betreten
  INSTALL /lib/modules/5.19.7-arch1-1/extra/xmm7360.ko
  SIGN    /lib/modules/5.19.7-arch1-1/extra/xmm7360.ko
At main.c:167:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  ZSTD    /lib/modules/5.19.7-arch1-1/extra/xmm7360.ko.zst
  DEPMOD  /lib/modules/5.19.7-arch1-1
make[1]: Verzeichnis „/usr/lib/modules/5.19.7-arch1-1/build“ wird verlassen
```

JoelLinn, 2022-09-08:
For me doing

```
sudo rmmod iosm
make load
```
Alway works.
Getting connected with the script may beed multiple tries but the driver is loading and creating the dev node fine

yummy527, 2022-09-08:
> For me doing
> 
> ```
> sudo rmmod iosm
> make load
> ```
> 
> Alway works. Getting connected with the script may beed multiple tries but the driver is loading and creating the dev node fine
Not work for me right now.

Where the module is installed by the command "sudo insmod xmm7360"? I suspect that the module is not properly installed on my machine.

Before trying this drive of pci version, I installed an [usb version](https://github.com/juhovh/xmm7360_usb). And I found that [the module generated by the usb version can not be removed completely at the moment](https://github.com/juhovh/xmm7360_usb/issues/20). Oops. Anyone can help?

yummy527, 2022-09-09:
Finally, xmm7360 module is successfully installed. Now I have a new problem to deal with:

```shell
archcarbon% lte up
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: ctlte
Script: /home/xxxx/Downloads/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x2, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x2
FCC lock: state 1 mode 2
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc02040000000200000000000000000000000000000000000000000 [... cut ...]
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0xef, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa0ed, 0x30209f, 0x0, 0x1, 0x1cc, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x30209f, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0xef, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa0ed, 0x30209f, 0x0, 0x1, 0x1cc, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
```

ch4m30, 2022-10-12:
I also have been experiencing the similar issue mentioned here, and followed the 
`sudo rmod iosm` command and the `make load` command that seemed to allow some progression. I was then successfully able to run `sudo ip link set wwan0 up` which has resulted in the following: 
`wwan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500` 

I am unsure exactly if I am to return to attempting to run a script or update a file in order to gain an internet connection from my sim; as I certainly feel as though I am missing a step.

Context: Garuda on Lenovo P53.

Thanks! No doubt the issue is the keyboard operator! ;)

PR #181 (2022-08-12): fix build with kernels >= 5.18.0
The `pci-dma-compat.h` API has been removed since linux 7968778914e53788a01c2dee2692cab157de9ac0
The "new" api is available for about 9 years now.

Fixes #179 
Although one has to blacklist iosm (or do `rmmod iosm`) because it otherwise reserves the xmm7360 card since 5.18. (Thanks @regenman)