xmm7360/xmm7360-pci #182: Install fails on Ubuntu 22

SmileSydney, 2022-05-31:
Installation on X1 Carbon Gen 6 with Fibocomm fails.

`**lspci**
.
03:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
.

**lsb_release -a**
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

**uname -r**
5.15.0-33-generic

**make && make load**
make -C /lib/modules/5.15.0-33-generic/build M=/home/ylee/tmp/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-33-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-33-generic'
sudo /sbin/rmmod xmm7360
**rmmod: ERROR: Module xmm7360 is not currently loaded**
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Operation not permitted
make: *** [Makefile:19: load] Error 1
`

sybnex, 2022-06-01:
Same here on HP Elitebook x360  :(

SmileSydney, 2022-06-02:
`test`

closed 2022-06-02 by SmileSydney