xmm7360/xmm7360-pci #183: Module does not compile on kernel 5.18

mithnae, 2022-06-01:
Due to braking changes in 5.18 in DMA handling, compilations ends with error
```
make -C /lib/modules/5.18.1-1.el8.elrepo.x86_64/build M=/home/mithnae/build/xmm7360-pci modules
make[1]: Entering directory '/usr/src/kernels/5.18.1-1.el8.elrepo.x86_64'
  CC [M]  /home/mithnae/build/xmm7360-pci/xmm7360.o
/home/mithnae/build/xmm7360-pci/xmm7360.c: In function ‘xmm7360_probe’:
/home/mithnae/build/xmm7360-pci/xmm7360.c:1451:8: error: implicit declaration of function ‘pci_set_dma_mask’; did you mean ‘pci_set_drvdata’? [-Werror=implicit-function-declaration]
  ret = pci_set_dma_mask(dev, 0xffffffffffffffff);
        ^~~~~~~~~~~~~~~~
        pci_set_drvdata
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /home/mithnae/build/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1834: /home/mithnae/build/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.18.1-1.el8.elrepo.x86_64'
make: *** [Makefile:9: default] Error 2
```

JoelLinn, 2022-06-02:
Dublicate of #179

JoelLinn, 2022-08-12:
Please check with current master. (Mind you may need to do rmmod iosm to test it)