xmm7360/xmm7360-pci #190: ERROR:root:Attach failed again, giving up

Oualid33, 2022-08-02:
The driver worked perfectly before I updated my Ubuntu 22.04 on 08/28/2022.
I got the following error: 

'bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x2, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x7aab3134
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c0204000001921100010102041100010102045e9ce009'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc02040000000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0xffffffff, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0x65, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa29, 0x92989, 0x0, 0x2, 0x25b, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x92989, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0x65, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa29, 0x92989, 0x0, 0x2, 0x25b, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
'

Oualid33, 2022-08-02:
My kernel is 5.15.0-43-generic on Thinkpad T14 Intel.

pswiatki, 2022-08-02:
You are telling tales, my friend. They are from the future ;-)

Oualid33, 2022-08-02:
> You are telling tales, my friend. They are from the future ;-)

Is it a bug?

pswiatki, 2022-08-02:
Yup. It was. Now it properly makes sense :)

yummy527, 2022-09-09:
> You are telling tales, my friend. They are from the future ;-)

why say that?

Oualid33, 2022-09-09:
@yummy527 I've made some grammatical and typographical errors then corrected them. So he commented in between

yummy527, 2022-09-09:
> @yummy527 I've made some grammatical and typographical errors then corrected them. So he commented in between

So, did your 4G module works well now?

pswiatki, 2022-09-09:
Yeah, it does not make sense for my comments to remain here (the date was from the future - that is all it was to it) because the original ones from @yummy527 were removed. So I removed mine as well. Hope the confusion is cleared now.

pswiatki, 2022-09-09:
By the way: I also noticed issues with driver's operation. Will check if they are similar to the one described here.

yummy527, 2022-09-09:
> By the way: I also noticed issues with the driver's operation. I will check if they are similar to the one described here.
You are nice.
I am not sure what exactly caused the error, but I suspect that it is related to the setting of apn, proxy, server name, user name, password and port. How can we send all these parameters to the xmm7360 module?

pswiatki, 2022-09-09:
I responded with my current problem here: #192 
To your question here: I guess all these parameters could be set up in `xmm7360.ini`

Oualid33, 2022-09-09:
> > @yummy527 I've made some grammatical and typographical errors then corrected them. So he commented in between
> 
> So, did your 4G module works well now?

Yeah, without doing anything but an update of my Ubuntu !

closed 2022-09-09 by Oualid33