xmm7360/xmm7360-pci #192: on thinkpad x1 carbon gen8, ERROR:root:Attach failed again

yummy527, 2022-09-09:
No PIN code on my sim card.
How to solve this problem? 

```shell
archcarbon% lte up
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: ctlte
Script: /home/xxxx/Downloads/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x2, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x2
FCC lock: state 1 mode 2
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000 [... cut ...]
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0xef, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa0ed, 0x30209f, 0x0, 0x1, 0x1cc, 0x0
unsolicited: UtaMsNetCellChangeIndCb: 0x0, 0x30209f, 0x0
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0xef, 0x1, 0x1, 0x0, 0x0, 0x0, 0xa0ed, 0x30209f, 0x0, 0x1, 0x1cc, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
```

pswiatki, 2022-09-09:
I started getting something similar:

```
[...]
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready

[...]

unsolicited: UtaMsSimCardIndCb: 0x0, 0x15, 0xff, 0x1, 0x3, 0x0, bytearray(XXXXXXXXXXXXXXX), 0x0
unsolicited: UtaMsSimApplicationIndCb: 0x0, 0x44, bytearray(XXXXXXXXXXXXXXXXXXXX), 0x10, 0x0, 0x0, bytearray(XXXXXXXXXXXXXXXXXXXXXX), 0x0
unsolicited: UtaMsSimInfoIndCb: 0x0, 0x0, 0x0, 0x0, 0x2fe2, bytearray(b'pl'), bytearray(b'en'), bytearray(b'\xff\xff'), bytearray(XXXXXXXXXXXX), 0xff, 0xa, 0xa, 0x3, 0x3, 0x1, 0x2, 0xff, 0x0, bytearray(XXXXXXXXXXXXXXX), 0x3, 0x0
unsolicited: UtaMsSimMccMncIndCb: 0x0, 0x0, 0x3, 0x104, 0x0
unsolicited: UtaMsSimLockStatusIndCb: 0x0, 0x8, bytearray(XXXXXXXXXXXXXXXXXXXXXX
unsolicited: UtaMsSimFdnStateIndCb: 0x0, 0x1, 0x1, 0x0, 0x0
unsolicited: UtaMsSimBdnStateIndCb: 0x0, 0x0, 0x1, 0x0, 0x0
unsolicited: UtaMsSimInitIndCb: 0x0, 0x0, 0x0, 0x0, 0x6f07, 0x0, 0x1, bytearray(XXXXXXXXXXXXX), ..............
unsolicited: UtaMsSimTkProactiveCommandIndCb: 0x0, bytearray(XXXXXXXXXXXXXXXXXX), 0x...................
unsolicited: UtaMsSimFullAccessIndCb: 0x0
unsolicited: UtaMsSmsSimMsgCacheFinishedIndCb: 0x0, 0x0
```
Then it stops for a minute or so and then starts spitting unsolicited messages every ~30 seconds:
```
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x32ff, 0x25, 0x0, 0x90, 0x1, 0x1, 0x1, 0x1, 0x3, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x2, 0x1, 0x2, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x3, 0x1, 0x3, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x4, 0x1, 0x4, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x5, 0x1, 0x5, 0x1, 0x0, 0x0
.....
```
It used to work before, but not any more with:

`Linux sdr-aptv 5.19.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 05 Sep 2022 18:09:09 +0000 x86_64 GNU/Linux`

yummy527, 2022-09-09:
Wi-Fi is a good choice now.🤷‍♂️

pswiatki, 2022-09-09:
Unrelated. Stick to the WWAN subject here.

ankitsinghaniyaz, 2022-09-23:
I'm getting the same error on `Linux thinkpd 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

yummy527, 2023-04-07:
Today I tried again, and it still does not work. The same error.

The card has no SIM PIN. The card is a normal one with both data service and calling service. It works well on my iPhone.

OS info: Linux archcarbon 6.2.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 30 Mar 2023 14:51:14 +0000 x86_64 GNU/Linux

In the `xmm7360.ini` file, I just set APN as `ctnet` or `ctwap` or `ctlet`,  since I the card is provider is China Telecom.

Any help?

yummy527, 2023-04-07:
In the file [open_xdatachannel](https://github.com/xmm7360/xmm7360-pci/blob/107b6baa9f380a55111a5ce29b987ef64c43104b/rpc/open_xdatachannel.py):
```python
# Add DNS values to /etc/resolv.conf
if not cfg.noresolv:
    with open('/etc/resolv.conf', 'a') as resolv:
        resolv.write('\n# Added by xmm7360\n')
        for dns in dns_values['v4'] + dns_values['v6']:
            resolv.write('nameserver %s\n' % dns)
```

Together with the file [xmm7360.ini](https://github.com/xmm7360/xmm7360-pci/blob/107b6baa9f380a55111a5ce29b987ef64c43104b/xmm7360.ini.sample) :
```txt
# uncomment to not add DNS values to /etc/resolv.conf
#noresolv=True
```

From the content of the two file mentioned above, I guess that: when running the command in shell `lte.sh setup` and `lte up`, the script is intended to write nameserver to the file `/etc/resolv.conf`. However, I just found that this conf file could not be changed by user and the lte script easily. Even after running command `echo "namesever 8.8.8.8" | sudo tee -a /etc/resolv.conf`(this command is mentioned in [INSTALLING.md](https://github.com/xmm7360/xmm7360-pci/blob/master/INSTALLING.md)), nothing was appended to file `resolv.conf` at all. 

[An old post](https://unix.stackexchange.com/questions/128220/how-do-i-set-my-dns-when-resolv-conf-is-being-overwritten) discussed how to set DNS in details, and I have not dig it a lot yet. But if this DNS writing problem causes the **Attach failed again, giving up** error mentioned here, the post will be helpful.