xmm7360/xmm7360-pci #194: /dev/xmm0/rpc error with kernel60 and kernel519

k1ngf15h3r, 2022-10-19:
I have installed majaro with kernel515 and successfully got. my xmm7360 Modem working with this repo. 

Testing new Kernels like 5.15 and 6.0 did not work. 

Got error missing /dev/xmm0/rpc

ch4m30, 2022-10-20:
Thanks for that. I am struggling to get it working with Garuda on 6.0.1-zen so may need to see if a different kernel will allow functionality.

k1ngf15h3r, 2022-10-25:
Today I tried it with kernel 6.0.3 in latest kubuntu 22.10 Bit without success. 

Bit I recognized a new mobile connection icon in the network manager

pswiatki, 2022-10-25:
Oh yeah? It would be great if you let us know if it worked for you (that new connection presented in the network manager).

xentity, 2022-11-16:
Same here (HP Elitebook 840 G5, Ubuntu 22.10):

```
Traceback (most recent call last):
  File "/home/dennis/Downloads/xmm7360/xmm7360-pci/./rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/dennis/Downloads/xmm7360/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
```
I have also an interface named wwan0, but if I want to use it in the Gnome settings, it says "no SIM found".

lluchs, 2022-11-30:
Looks like there is now built-in support for the 7360 chip by Intel in the kernel: https://github.com/torvalds/linux/commit/1f52d7b622854b8bd7a1be3de095ca2e1f77098e

Looking at the [list of recent commits](https://github.com/torvalds/linux/commits/master/drivers/net/wwan/iosm) there's a bunch of bugfixes that probably haven't made it into the distribution kernels yet. An interesting one is the intel_iommu fix - setting `intel_iommu=on` gave me some additional error output, but made it disappear from ModemManager.

```
~ dmesg | grep iosm
[   20.401542] iosm 0000:02:00.0: enabling device (0000 -> 0002)
[  129.961643] iosm 0000:02:00.0: PORT open refused, phase A-CD_READY
[  129.962920] iosm 0000:02:00.0: PORT open refused, phase A-ROM
[  129.963491] iosm 0000:02:00.0: PORT open refused, phase A-ROM
[  129.963984] iosm 0000:02:00.0: PORT open refused, phase A-ROM
```

Anyone else having more luck with iosm?

lluchs, 2022-11-30:
Anyways, the issue here (xmm7360 not working on newer kernels) is easily solved by unloading iosm before loading xmm7360.

ch4m30, 2022-12-31:
> Anyways, the issue here (xmm7360 not working on newer kernels) is easily solved by unloading iosm before loading xmm7360.

I am confused by this sorry @lluchs - how is that "unloaded"? I looked at the referenced link but struggled to understand how to execute it. Sorry.

lxs602, 2023-02-09:
> lluchs

```rmmod iosm
...does not make any difference. Is there anything else that needs to be done?

lxs602, 2023-02-09:
Unloading iosm with `rmmod iosm` does not help me on Ubuntu 22.04 with kernel 6.0.19. I have the same problems as xenity above.

lxs602, 2023-02-09:
See also the [arch wiki](https://wiki.archlinux.org/title/Xmm7360-pci)

Linux Kernel support and Modem Manager integration
Support for this device has been added since Linux 5.18 (see [torvalds/linux@1f52d7b](https://github.com/torvalds/linux/commit/1f52d7b622854b8bd7a1be3de095ca2e1f77098e)). ModemManager accesses such WWAN modems via MBIM interface, provided by the iosm kernel module. The GL-860 does provide a MBIM interface, the GL-850 does not. Fibocom/Intel don't seem to care, there is no announcement/rumor that they will be implementing it.

That is why [xmm7360-pci/issue/31](https://github.com/xmm7360/xmm7360-pci/issues/31) correctly states that even with the kernel driver, the modem doesn't work: With iosm kernel module loaded, the interface shows up as /dev/wwan0at0 and /dev/wwan0at1, and is seen by mmcli -L, but journalctl yields SIM not inserted:

ModemManager:   <info>  [device /sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0]
                        creating modem with plugin 'Intel' and '3' ports
ModemManager:   <warn>  [plugin/intel] could not grab port wwan0at0:
                        Cannot add port 'wwan/wwan0at0', unhandled port type
ModemManager:   <info>  [base-manager] modem for device '/sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0'
                        successfully created
ModemManager:   <warn>  [modem3] couldn't load supported IP families: SIM not inserted
ModemManager:   <info>  [modem3] state changed (unknown -> locked)
ModemManager:   <warn>  [modem3] modem couldn't be initialized: Couldn't check unlock status: SIM not inserted
ModemManager:   <info>  [modem3] state changed (locked -> failed)
ModemManager:   <warn>  [modem3] error initializing: Modem in failed state: sim-missing
ModemManager:   <info>  [1673136941.3379] manager: (wwan0at1): new Broadband device
                        (/org/freedesktop/NetworkManager/Devices/11)
NetworkManager: <info>  [1673136941.3382] device (wwan0at1): state change: unmanaged -> unavailable
                        (reason 'managed', sys-iface-state: 'external')
NetworkManager: <info>  [1673136941.3385] device (wwan0at1): modem state 'failed'
NetworkManager: <info>  [1673136941.3387] modem-broadband[wwan0at1]:
                        failed to retrieve SIM object: No SIM object available
There is ongoing development in [ModemManager/issue/612](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/612): A patch was accepted in Linux 6.2rc1 that adds a low-level mbim.rpc interface intended to ultimately be used as [kernel module](https://github.com/tao-j/iosm-rpc-dkms) which shall be script-glued into ModemManager (and be compatible to the xmm7360-pci python3 rpc/open_xdatachannel.py script).

yummy527, 2023-04-08:
> Same here (HP Elitebook 840 G5, Ubuntu 22.10):
>
> ```
> Traceback (most recent call last):
>   File "/home/dennis/Downloads/xmm7360/xmm7360-pci/./rpc/open_xdatachannel.py", line 42, in <module>
>     r = [...]

As @lluchs suggested, this error could be solved by unloading iosm.
```
sudo rmmod iosm
```

And then try again.

PR #199 (2023-07-17): Unload iosm module before loading xmm7360
The built-in iosm module now supports the 7360 chip (in theory) and will automatically bind to it. This prevents xmm7360 from binding to the device.

Fixes #194
closed 2023-07-17 by tgxn