xmm7360/xmm7360-pci #196: Lenovo T14s Gen1 debian

toxicbiocide, 2022-10-26:
hi, having issues while installing the driver. Appreciate your support.
> dependencies installed
> **INSTALLING.md** instructions followed step by step
```sudo ./scripts/lte.sh setup``` result in error:

```
toxic968@debian:~/Downloads/xmm7360-pci$ sudo ./scripts/lte.sh setup
lte.sh: manage xmm7360-pci
APN: wwan0
Script: /home/toxic968/Downloads/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

Requirement already satisfied: pyroute2 in /root/.local/lib/python3.9/site-packages (0.7.3)
Requirement already satisfied: configargparse in /root/.local/lib/python3.9/site-packages (1.5.3)
make -C /lib/modules/5.10.0-19-amd64/build M=/home/toxic968/Downloads/xmm7360-pci modules
make[1]: *** /lib/modules/5.10.0-19-amd64/build: Datei oder Verzeichnis nicht gefunden.  Schluss.
make: *** [Makefile:9: default] Fehler 2
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Fehler 1 (ignoriert)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not load module xmm7360.ko: No such file or directory
make: *** [Makefile:19: load] Fehler 1
```

gilou, 2023-06-16:
Hi, I'd say you're missing the kernel headers maybe? (yes, you probably found a solution by now…)

nursoda, 2023-09-02:
@toxicbiocide, please close this issue.

toxicbiocide, 2023-09-08:
problem parked - other things atm

closed 2023-09-08 by toxicbiocide