xmm7360/xmm7360-pci #200: Feature request: add rmmod isom to lte.sh

RealTehreal, 2023-01-09:
Greets,

first, thanks for still keeping this project up and running for so long already!

Now to topic: I'd like to request adding `rmmod iosm` to the `up` area within `lte` script. I think this would save some non-technical persons some trouble finding out, why the project won't compile / run.

Keep up the work and stay save.

Greets

RealTehreal, 2023-01-10:
Added a pull request for this.
[https://github.com/xmm7360/xmm7360-pci/pull/201](https://github.com/xmm7360/xmm7360-pci/pull/201)

yummy527, 2023-04-08:
Good request. 

I remember that it took me a lot time to find the **secret** that iosm blocks xmm7360. It is an possible obstacle to some new guys.