xmm7360/xmm7360-pci #204: It partially works on x1 carbon gen8 with archlinux 6.2.9

yummy527, 2023-04-07:
Hi,guys. 
Finally, it works after struggling. Here I just give some steps to reproduce how to make it work.

# machine and OS
machine: Lenovo X1 carbon gen8
OS: archlinux with kernel 6.2.9
Python: 3.10(I am not sure whether it maters or not.)

# steps to reproduce successful work
step 1: After a **fresh reboot**, just "remove" or blacklist the iosm drive with command: 
```shell
sudo rmmod iosm
```
**This step is important!** If iosm is not removed, it will `block` the xmm7360-pci drive and you will the warning like "insert a sim card to use this modem".

*This command above will not delete iosm from your disk. Every time you boot your machine, the iosm will be there because it is in the kernel. You can check it with command `lspci -k` and if you carefully check the output, you will find iosm there.*

step 2: Install ModemManager
If you are using ArchLinux as me,
```shell
sudo pacman -S modemmanager
```
If you are using Ubuntu, 
```shell
sudo apt-get update
sudo apt-get -y install modemmanager
```
If other OS, just find the right way to install ModemManager.

step 3: edit  `xmm7360.ini`
Open the file `xmm7360.ini`, and make sure to
- set your apn. (I am not sure yet, but it seems that wrong apn will also work. You can select APN in the setting pannel later.) 
- **uncomment `dbus=True`**

Here is mine:
```txt
# driver config
apn="ctnet"

#ip-fetch-timeout=1
#nodefaultroute=True
#metric=1000

# uncomment to not add DNS values to /etc/resolv.conf
#noresolv=True

# used to activate NetworkManager integration
dbus=True

# Setup script config
BIN_DIR=/usr/local/bin

```
step 4: execute the script `lte.sh`
```shell
cd ~/Downloads/xmm7360-pci # cd to where your downloaded folder `xmm7360-pci` is.
sudo ./scripts/lte.sh setup
```
That's all for the script. Command `lte up` is not needed here.

# Others
I find this working method somewhat occasionally. The "giving up" error stops me for more than half a year (cry...). In these several days, I just wonder whether the LTE or 4G could work or not on my Linux laptop.

After searching and trying many pages on the web, finally I find this one works.

If the **Mobile Network** in the system setting panel just tells you "**insert a sim card to use this modem**" even if you have already insert you card. The trick is: repeat step 2 and step 4. That is reinstall ModemManager and run `lte.sh` script again.

If the **Mobile Network** in the system setting panel disappears(yes it may disappear when you running the lte.sh), as well reinstall ModemManager and run `lte.sh` script again. 

```shell
sudo pacman -S modemmanager
cd ~/Downloads/xmm7360-pci # cd to where your downloaded folder `xmm7360-pci` is.
sudo ./scripts/lte.sh setup
```
I have no idea why merely trying `sudo ./scripts/lte.sh setup` does not work. But **reinstalling modemmanager and then running the script WORKS.**

That's all my finding. Hope it works for you. 
[image]

It is a little strange that the signal shows very weak, 0%. But downloading and uploading works. 

I am excited that it finally works. If finding other `techniques` later, I will share here.

dieselburner, 2023-04-12:
> The trick is to repeat step 2 and step 4. That is to reinstall ModemManager and run lte.sh script again.

No. That's not how open source supposed to work.

yummy527, 2023-04-12:
> > The trick is to repeat step 2 and step 4. That is to reinstall ModemManager and run lte.sh script again.
> 
> No. That's not how open source supposed to work.

Apologize. Reinstalling ModemManager is not needed at all.

rafaelrsanches, 2023-04-12:
My output of `lspci -k` on Fedora looks like this:

`05:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
	Subsystem: Intel Corporation Device 0020
	Kernel modules: iosm, xmm7360`

yummy527, 2023-04-12:
> My output of `lspci -k` on Fedora looks like this:
> 
> 
> 
> ```
> 
> 05:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
> 
> 	Subsystem: Intel Corporation Device 0020
> 
> 	Kernel modules: iosm, xmm7360
> 
> ```
> 
> 
> 
> 

In your case, it seems that iosm is "removed" and xmm7360 is installed but not used.

cd to the downloaded xmm7360 folder, and try step 4. That is my suggestion.

rafaelrsanches, 2023-04-12:
> In your case, it seems that iosm is "removed" and xmm7360 is installed but not used.
>
> cd to the downloaded xmm7360 folder, and try step 4. That is my suggestion.

I'm able to do step 4, but the Kernel module doesn't go to "Kernel driver in use:". And if I run `sudo lte up` I got the following error:
```
lte.sh: manage xmm7360-pci
APN: myapn.com
Script: /home/user/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
Traceback (most recent call last):
  File "/home/user/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
        ^^^^^^^^^^^^
  File "/home/user/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
Cannot find device "wwan0"
``

rafaelrsanches, 2023-04-18:
I was able to get it working on Fedora 38 (with Secure Boot disabled), but I have 2 questions.

1. You need to do this steps every time you reboot?
2. Is it possible to do this with Secure Boot enabled?

tgxn, 2023-07-17:
I merged the rmmod ioctl thing today, should fix this.

KataFF, 2023-08-19:
I tried the steps described in the OP, but it did not work for me (see #209)

vebodev, 2023-11-20:
yummy527, 

I did the same instructions on Archlinux. However I met a different case not mentioned before.
[image]
It is not the same UI as yours. I have no idea to find out the same config as yours.

Please advice what should I do. thank you

yummy527, 2024-01-24:
> yummy527,
>
> I did the same instructions on Archlinux. However I met a different case not mentioned before. [...]

As far I could remember,  it just worked after second `sudo ./scripts/lte.sh setup`. I changed nothing related to user name and password in your screenshot, just using the default ones. Be sure to set your APN in `xmm7360.ini`. Since you are using China Telecom, just `apn="ctnet"`.

PS: Long time no use this xmm7360 tool. Today I tried to run it, and just found that it didn't work now.

SimPilotAdamT, 2024-02-22:
I seem to have some success... I [forked this project](https://github.com/SimPilotAdamT/xmm7360-pci-SPAT) and managed to get it compiling after some fiddling, and the module works. My only remaining issue is integration with ModemManager, which seems to fail on my device... I am using a ThinkaPad P52s with `lspci` output:
```
03:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem [8086:7360] (rev 01)
        Subsystem: Intel Corporation XMM7360 LTE Advanced Modem [8086:0020]
        Flags: bus master, fast devsel, latency 0, IRQ 150
        Memory at dd200000 (64-bit, non-prefetchable) [size=4K]
        Memory at dd201000 (64-bit, non-prefetchable) [size=1K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/4 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Latency Tolerance Reporting
        Capabilities: [150] L1 PM Substates
        Kernel driver in use: xmm7360
        Kernel modules: iosm
```
I am using Arch Linux with the latest Arch kernel in the stable repos. Having followed the steps in your original post, ModemManager refuses to pick up my module (I have to use the `lte` commands instead), spitting these into the journald log:
```
Feb 22 15:25:26 adam-arch-laptop systemd[1]: Starting Modem Manager...
Feb 22 15:25:26 adam-arch-laptop ModemManager[7776]: <msg> ModemManager (version 1.22.0-1) starting in system bus...
Feb 22 15:25:26 adam-arch-laptop systemd[1]: Started Modem Manager.
Feb 22 15:25:30 adam-arch-laptop ModemManager[7776]: <msg> [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1c.6/0000:04:00.0': not supported by any plugin
Feb 22 15:25:30 adam-arch-laptop ModemManager[7776]: <msg> [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:1f.6': not supported by any plugin
```