xmm7360/xmm7360-pci #205: ThinkPad X390   Fedora 38    An error is reported when executing sudo python3 rpc/open_xdatachannel.py

Perseus-Cluster, 2023-05-06:
设备：ThinkPad X390 20SD
模块：广和通L850-GL 
系统：Fedora 38
问题：
在执行sudo python3 rpc/open_xdatachannel.py报错
错误提示：
File "/root/tmp/xmm7360-pci/rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
        ^^^^^^^^^^^^
  File "/root/tmp/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'

另外依赖的安装参考91号反馈中提供的依赖。

附图：
[image]

Device: ThinkPad X390 20SD
Module: Fibocom L850-GL
System: Fedora 38
question:
An error is reported when executing sudo python3 rpc/open_xdatachannel.py
Error message:
File "/root/tmp/xmm7360-pci/rpc/open_xdatachannel.py", line 42, in <module>
     r = rpc.XMMRPC()
         ^^^^^^^^^^^^^
   File "/root/tmp/xmm7360-pci/rpc/rpc.py", line 19, in __init__
     self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'

For the installation of other dependencies, refer to the dependencies provided in Feedback 91.

Nannk, 2023-05-06:
did you `lte setup` aka `make && make load` ?

jklaus, 2023-05-16:
I have the same problem with Fedora 37 on a Lenovo P15s.