xmm7360/xmm7360-pci #211: How-to: Vanilla Linux Kernel 6.2

Mixaill, 2023-09-25:
0. Check that you have Linux 6.2 or greater
1. use https://github.com/xmm7360/xmm7360-pci/pull/210 or change the XMM RPC path to the `/dev/wwan0xmmrpc0` inside the `rpc/rpc.py`
2. rename `xmm7360.ini.sample` to `xmm7360.ini` 
3. execute `sudo scripts/lte.sh up`
4. it works!

pswiatki, 2023-11-09:
Everything nicely connected, but this is the end of `./lte up`:
```
name:HotSpot IC uuid:b318a583-3337-4987-9dee-2202e051df07 type:802-11-wireless
name:lo uuid:75734424-8b13-4775-a325-9958ccc90d26 type:loopback
setup 4e34f700-f82e-490a-97fb-9670d84d2e66
setup connection
Traceback (most recent call last):
  File "/home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 240, in <module>
    manager.ActivateConnection(connection_path, devpath, "/")
```
What is wrong here?

pswiatki, 2023-11-09:
Oh, by the way: it seems `systemd-homed` service shall be started or no DNS info gets stored in `/etc/resolv.conf`.
Is this true, or I have something messed up on my system?

pswiatki, 2023-11-09:
> Traceback (most recent call last):
>   File "/home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 240, in <module>
>     manager.ActivateConnection(connection_path, devpath, "/")
>                                                 ^^^^^^^
> NameError: name 'devpath' is not defined
> ```
> 
> What is wrong here?

Hmmm.... when looking at this code:

```
for d in devices:
    dev_proxy = system_bus.get_object("org.freedesktop.NetworkManager", d)
    prop_iface = dbus.Interface(dev_proxy, "org.freedesktop.DBus.Properties")
    props = prop_iface.GetAll("org.freedesktop.NetworkManager.Device")
    if props["Interface"] == "wwan0":
        devpath = d
        print("found Interface: %s" % props["Interface"])
        print("Managed: %s" % props["Managed"])
        if props["Managed"] == 0:
            print("activate")
            prop_iface.Set("org.freedesktop.NetworkManager.Device",
                           "Managed", dbus.Boolean(1))
```
I realized it may be failing because `NetworkManager` somehow doesn't know about the newly brought up `wwan0`. Can this be? If this is true,  `devpath` will not be set.

pswiatki, 2023-11-09:
Another obseravtion: it is probably `iosm` that doesn't like hibernation, though:
```
Nov 09 16:49:49 sdr-aptv sudo[18382]: pswiatki : TTY=pts/1 ; PWD=/home/pswiatki/devel/HW/LTE_modem_m.2/xmm7360-pci/scripts ; USER=root ; COMMAND=./lte.sh up
Nov 09 16:49:49 sdr-aptv sudo[18382]: pam_unix(sudo:session): session opened for user root(uid=0) by pswiatki(uid=1000)
Nov 09 16:49:49 sdr-aptv kernel: iosm 0000:73:00.0: PORT open refused, phase A-CD_READY
Nov 09 16:49:50 sdr-aptv kernel: iosm 0000:73:00.0: msg timeout
Nov 09 16:49:50 sdr-aptv kernel: iosm 0000:73:00.0: msg timeout
Nov 09 16:49:51 sdr-aptv sudo[18382]: pam_unix(sudo:session): session closed for user root
Nov 09 16:49:51 sdr-aptv kernel: iosm 0000:73:00.0: ch[0] timeout
Nov 09 16:49:51 sdr-aptv kernel: iosm 0000:73:00.0: if_id 0: OPEN_SESSION send failed
Nov 09 16:49:51 sdr-aptv kernel: iosm 0000:73:00.0: cannot connect wwan0 & id 0 to the IPC mem layer
```
So, not possible to bring the `wwan0` interface back up after coming back from hibernation.

adi1, 2023-12-21:
> I realized it may be failing because `NetworkManager` somehow doesn't know about the newly brought up `wwan0`. Can this be? If this is true (i.e. NM doesn't know about `wwan0`), `devpath` will never be set and the script will conclude ungracefully.

Were you able to get anywhere with this? Everything works for me except for the NetworkManager integration. I've also tried the ModemManager 1.20.4 included with my Debian 12.4 system, but it can't connect, although it does show the modem and the provider. I've also tried the forked version mentioned above, but that gives the same result.

So right now, I can run `lte up`, ignore the `devpath` error, have no connection information in NetworkManager, run `lte down` when I'm done with the internet.