xmm7360/xmm7360-pci #212: L850-GL only works once on Lenovo T480 Linux Debian, have to restart PC if disconnected

hoangthisd, 2023-10-05:
Hello,
I followed the tutorial:
$ git clone https://github.com/xmm7360/xmm7360-pci.git
$ cd xmm7360-pci
$ make && make load
# python3 rpc/open_xdatachannel.py --apn apn.url
# echo "nameserver 1.1.1.1" >> /etc/resolv.conf
# ip link set wwan0 up

At first, It works like a charm, but after a while, the network disconnects, and I cannot make it reconnect. INFO:root:IP address and DNS server show the correct IP, but the response is: 0xffffffff

All I can do is to restart the machine, and then it works again.

PaddyAB, 2023-10-29:
I had 2 issues like this.. (1) how long does it run for? have had a couple of systems it runs for 24hrs and then hangs up..  This was traked back to an ISP log lock issue were the ISP puts a split sec pause so it can roll over the log and then carries on but the modem falls over.. Fix for this was a timer on the power so the router gets powered down once per day at 2am in the morning..

(2) royal pain anything happens and the only way to get things working again is power cycle to get the modem back up.. Didnt helps the firmware was also crashing often.. I managed to get it fixed by recompiling the MHI module.. Not sure what your config is or if is possible on with what you are working with.. I was working on full linux disto Details so if of interest:

The change is to change one place from sync to async
+++ b/drivers/bus/mhi/pci_generic.c
@@ -637,7 +637,7 @@ static void mhi_pci_recovery_work(struct work_struct *work)
if (err)
goto err_try_reset;

- err = mhi_sync_power_up(mhi_cntrl);
+ err = mhi_async_power_up(mhi_cntrl);
if (err)
goto err_unprepare;

make sure to find the correct entry as there is another similar..

/* MHI bus does not power up the controller by default */
err = mhi_prepare_for_power_up(mhi_cntrl);
if (err) {
dev_err(&pdev->dev, "failed to prepare MHI controller\n");
goto err_unregister;
}

err = mhi_async_power_up(mhi_cntrl);
if (err) {
dev_err(&pdev->dev, "failed to power up MHI controller\n");
goto err_unprepare;

janicez, 2025-09-17:
While I realize this is not the place to talk about the iosm driver, I am having exactly the same problem with both xmm7360-pci on kernel versions which support it (too new a kernel version and the driver doesn't work), and iosm on kernel versions where it supports the XMM7360. I wonder how on Windows the modem is told to hang up and come back around, or indeed how it tells the driver «hey, I've been hung up on, report hang-up status to the OS»

On xmm7360-pci versions I have been able to get the modem working again by bouncing it using /sys files, but that disables the C6 power state, noticeably increasing my computer's power consumption.