xmm7360/xmm7360-pci #218: Sim not found in Ubuntu 22.04 LTS enevo ThinkPad T490s

asadlive84, 2024-03-05:
My Sim is working for windows 10 but not working Ubuntu 22.04
My Device is Lenovo ThinkPad T490s
 -----------------------------
  General  |              path: /org/freedesktop/ModemManager1/Modem/0
           |         device id: b9c6aa7a36498de65208bbbca1da2cf1f6d89309
  -----------------------------
  Hardware |      manufacturer: Fibocom
           |             model: L850 LTE Module","L850
           | firmware revision: 18500.5001.00.05.27.30
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 015550002692173
  -----------------------------
  System   |            device: /sys/devices/pci0000:00/0000:00:1c.6/0000:3b:00.0
           |           drivers: iosm
           |            plugin: Intel
           |      primary port: wwan0at1
           |             ports: wwan0 (net), wwan0at1 (at)
  -----------------------------
  Status   |             state: failed
           |     failed reason: sim-missing
           |       power state: low
  -----------------------------
  Modes    |         supported: allowed: 3g, 4g; preferred: none
           |           current: allowed: 3g, 4g; preferred: none

quytttb, 2024-03-17:
You should install older version (20.04)

alexnetdoc, 2024-04-05:
Do you guys know when the fix will be available for 22.04? No sim error is a pain...

pswiatki, 2024-04-05:
Sadly, it will never be available.

xmha97, 2025-05-28:
[image]
My Sim is working for windows 11 but not working Ubuntu 25.04

pswiatki, 2025-05-31:
Get familiar with [this Merge Request](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1280) and check other MRs mentioned in it, too.

xmha97, 2025-06-01:
> Shure thing!
> Get familiar with [this Merge Request](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1280) and check other MRs mentioned in it, too.

I'm confused. Can you please just give me a few commands?

pswiatki, 2025-06-02:
I posit you must build a specific branch of `ModemManager` to be able to use `xmm7360`.