xmm7360/xmm7360-pci #219: Compilation Failure on Kernel 6.8.5-zen1-1-zen for xmm7360-pci on TP X1 Yoga Carbon Gen 4 running Garuda

aarek-eng, 2024-04-13:
### Description:
I attempted to install the `xmm7360-pci` driver on Garuda Linux with the `6.8.5-zen1-1-zen` kernel using the AUR package `xmm7360-pci-git`. The installation process failed during the DKMS build step due to incompatible pointer types in the driver code.

### Environment:
- **Operating System:** Garuda Linux
- **Kernel Version:** 6.8.5-zen1-1-zen
- **Driver Version:** xmm7360-pci-git r247.a8ff2c6

### Steps to Reproduce:
1. Install the `xmm7360-pci-git` package from the AUR.
2. Run `sudo dkms install --no-depmod xmm7360-pci-git/r247.a8ff2c6 -k 6.8.5-zen1-1-zen`.

### Expected Behavior:
The driver should compile and install without errors.

### Actual Behavior:
The driver compilation fails with an error related to incompatible pointer types for the `xmm7360_tty_write` function.

### Error Log:
```
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1342:18: error: initialization of 'ssize_t (*)(struct tty_struct *, const u8 *, size_t)' from incompatible pointer type 'int (*)(struct tty_struct *, const unsigned char *, int)' [-Werror=incompatible-pointer-types]
 1342 |         .write = xmm7360_tty_write,
      |                  ^~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1342:18: note: (near initialization for 'xmm7360_tty_ops.write')
```

Full DKMS make log is reproduced below at the bottom of the ticket.

### Additional Information:
- The issue appears to be a mismatch in function pointer types, potentially due to changes in the kernel API that the driver has not yet adapted to.
- Warnings and additional build information are provided in the DKMS make log.

### Suggested Fix (if any):
Review the function signature for `xmm7360_tty_write` and update it to match the expected kernel interface for tty device operations.

---

```Cmake
File: /var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/make.log
DKMS make.log for xmm7360-pci-git-r247.a8ff2c6 for kernel 6.8.5-zen1-1-zen (x86_64)
2024-04-13T13:03:37 CEST
make -C /usr/lib/modules/6.8.5-zen1-1-zen/build M=/var/lib/dkms/xmm7360-pci-git/r247.a8ff2
c6/build modules
make[1]: Entering directory '/usr/lib/modules/6.8.5-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.o
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:656:5: warning: no previous pro
totype for 'xmm7360_cdev_open' [-Wmissing-prototypes]
  656 | int xmm7360_cdev_open(struct inode *inode, struct file *file)
      |     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:664:5: warning: no previous pro
totype for 'xmm7360_cdev_release' [-Wmissing-prototypes]
  664 | int xmm7360_cdev_release(struct inode *inode, struct file *file)
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:670:9: warning: no previous pro
totype for 'xmm7360_cdev_write' [-Wmissing-prototypes]
  670 | ssize_t xmm7360_cdev_write(struct file *file, const char __user *buf,
      |         ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:684:9: warning: no previous pro
totype for 'xmm7360_cdev_read' [-Wmissing-prototypes]
  684 | ssize_t xmm7360_cdev_read(struct file *file, char __user *buf, size_t size,
      |         ^~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c: In function 'xmm7360_irq0':
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1191:13: warning: the compariso
n will always evaluate as 'true' for the address of 'td_ring' will never be NULL [-Waddres
s]
 1191 |         if (xmm->td_ring) {
      |             ^~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:225:24: note: 'td_ring' declare
d here
  225 |         struct td_ring td_ring[16];
      |                        ^~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c: At top level:
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1342:18: error: initialization
of 'ssize_t (*)(struct tty_struct *, const u8 *, size_t)' {aka 'long int (*)(struct tty_st
ruct *, const unsigned char *, long unsigned int)'} from incompatible pointer type 'int (*
)(struct tty_struct *, const unsigned char *, int)' [-Werror=incompatible-pointer-types]
 1342 |         .write = xmm7360_tty_write,
      |                  ^~~~~~~~~~~~~~~~~
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1342:18: note: (near initializa
tion for 'xmm7360_tty_ops.write')
/var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build/xmm7360.c:1445:6: warning: no previous pr
ototype for 'xmm7360_dev_init_work' [-Wmissing-prototypes]
 1445 | void xmm7360_dev_init_work(struct work_struct *work)
      |      ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/xmm7360-pci-git/r247.a8ff2c6/build
/xmm7360.o] Error 1
make[2]: *** [/usr/lib/modules/6.8.5-zen1-1-zen/build/Makefile:1919: /var/lib/dkms/xmm7360
-pci-git/r247.a8ff2c6/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.8.5-zen1-1-zen/build'
make: *** [Makefile:9: default] Error 2
```

mfournier, 2024-04-23:
Hello @aarek-eng !

Can you give #220 a try on your system ?

Gowee, 2024-06-15:
3259b38e35ba72319c2c8a52ed57bccf7399c6e3 works for me on Arch Linux with the kernel `6.9.4-zen1-1-zen`.