xmm7360/xmm7360-pci #222: Current status on Fedora 40 (kernel version 6.8.8)

tuxor1337, 2024-05-04:
First of all, thanks for your work to all contributors in this repository!

I just wanted to report how I have been able to create an LTE internet connection on my Fedora 40 machine based on the code and knowledge stored in this repository (and pull requests).

# My exact setup

* ThinkPad X1 Carbon 7th generation (`20QDS06E00`)
* Fedora 40 with GNOME 46.1
* Linux kernel version: `6.8.8-300.fc40.x86_64`
* lspci: `02:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem [8086:7360] (rev 01)`
* Firmware version: `18500.5001.05.02.24.09`
* SIM card: O2-based provider in Germany, PIN deactivated
* the iosm module is blacklisted in `/etc/modprobe.d/00-iosm-blacklist.conf` (a single line: `blacklist iosm`)
* I bought the ThinkPad as a used device, there is no Windows installation on it (hence, I never used the WWAN card under Windows), but I don't know whether previous owners of the laptop have used or updated the WWAN card under Windows.

# PRs that I applied

* https://github.com/xmm7360/xmm7360-pci/pull/220: Without this patch, `make` will fail with the error message quoted in the PR.
* https://github.com/xmm7360/xmm7360-pci/pull/216: This patch was necessary to setup the modem as a NetworkManager device (via DBus).

# General setup

```shell
$ sudo dnf install python3-pyroute2 python3-configargparse make automake gcc gcc-c++ kernel-devel
$ git clone https://github.com/xmm7360/xmm7360-pci
$ cd xmm7360-pci
$ git remote add mfournier https://github.com/mfournier/xmm7360-pci
$ git remote add bill88t https://github.com/bill88t/xmm7360-pci
$ git fetch --all --prune
$ git checkout -b develop
$ git merge mfournier/linux-6.6
$ git merge bill88t/master
$ make && make load
$ cp xmm7360.ini.sample xmm7360.ini
$ # edit xmm7360.ini by adapting the following:
$ # insert your apn in this line: apn=your.apn.here
$ # uncomment this line: noresolv=True
$ sudo python rpc/open_xdatachannel.py
```

# Setting DNS name servers

In the output of the last command above, I found a line listing DNS entries, and I added them to `systemd-resolved` using the following command line (replace the IP addresses with your outputs):

```shell
$ resolvectl dns wwan0 62.109.121.18 62.109.121.17
```

Note that appending the DNS name servers to the `/etc/resolv.conf` as implemented in the `open_xdatachannel.py` script did not work for me!

# Register device with NetworkManager

These steps will establish a working internet connection on my setup. However, the connection is not registered with NetworkManager. To achieve this, uncomment the line `dbus=True` in your `xmm7360.ini` and rerun the `open_xdatachannel.py` script. After that, the output of `nmcli` confirms that a new connection has been established, like this:

```
wwan0: connected to xmm7360
	"Intel XMM7360 Advanced"
	unknown (xmm7360), hw, mtu 1500
	inet4 10.117.252.0/32
	route4 10.117.252.0/32 metric 950
	route4 default via 10.117.252.0 metric 950
	inet6 2a02:3030:604:fa0d:9359:b803:fda8:674d/64
	inet6 fe80::7ea2:849e:4d8e:8e80/64
	route6 fe80::/64 metric 256
	route6 default via fe80::1 metric 1024
```

Note that the connection is shown as "unknown" type. It does not show up in the NetworkManager GUI (using GNOME). The NetworkManager status icon in the GNOME panel will show the icon that is also used for a wired LAN connection. It would be interesting to have a way to have this show up as a mobile broadband connection. However, I guess that this would be of no use as long as it is not managed with ModemManager.

# Remaining steps (not Fedora-specific)

Following these steps, we can actually use this WWAN modem. However, it would be very much desirable to have the mainline Linux kernel and ModemManager pick up this work to seemlessly integrate the WWAN connection into the NetworkManager GUI. Here is my understanding why we are not there yet:

* One main issue is that the mainline kernel handles XMM 7360 and 7560 equally, even though 7360 uses a different "protocol" (AT instead of MBIM port). A kernel patch to fix this has been [submitted in September 2022](https://lore.kernel.org/all/20220926040524.4017-1-shaneparslow808@gmail.com/), but was never merged. Why hasn't it been merged? One reason seems to be that some people expected to see a firmware update for the Fibocom cards that would enable the MBIM port so that, in fact, the 7360 can be treated like a 7560 device. However, such a firmware update never came, and nobody came back to reconsider the kernel patch. Another reason might be that the modem can crash when the system is going to sleep, and the kernel maintainers requested that the kernel patch should fix this. I have no idea whether this is really strictly required for a merge, or how hard it would be to fix this.
* Even if the mentioned kernel patch would be merged, further changes to [ModemManager](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/612) would be required. Basically, the RPC-functionality implemented in the `open_xdatachannel.py` script from this repository needs to be cleaned up, and transferred to the ModemManager project.

pswiatki, 2024-05-04:
Very well, indeed. So, you are in essence saying the xmm7360 driver is no longer needed. Just the RPC interface, since the firmware from Fibocom does not support MM/NM preferred MBIM. We have been here already, but good summary, thanks.

tuxor1337, 2024-06-29:
Update: Based on the Python scripts in this repo, I created a merge request to add support for the XMM7360 in RPC mode to ModemManager: [!1200](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200) If anyone here has capabilities to test my implementation on their setup, please do and report about your experience in the MR. Thanks in advance!

pswiatki, 2024-06-30:
I tested it and commented in the MR. Works very well.
Thank you so much, @tuxor1337

ETH1166, 2024-08-15:
Thank you very much Thomas @tuxor1337 !
Following your tutorial for Fedora 40 I was able to connect to mobile broadband for the first time on Fedora's latest kernel 6.10.4. The only thing I needed to add is to sign the module for compatibility with my secure boot settings:
`sudo /lib/modules/$(uname -r)/build/scripts/sign-file sha512 /root/mok.priv /root/mok.der xmm7360.ko`
I have also tried to unlock my SIM with AT (AT+CPIN="<pin>") commands, but failed on this. Only after unlocking the SIM-card, I have been able to connect.
As a result I can announce, that the HP Elitebook x360 1030 G3 is also compatible with this driver.
I would love to see kernel support for this modem type, as it is quite scary to fire up a broadband connection through Python-scripts.

Next I will try to run kernel based using the `iosm` module through `/dev/wwan0xmmrpc0`. Did I get you right, that I do no longer need the module I have been struggeling to compile and sign?

If ModemManager would support this driver, I would be completely glad. I am unsure, whether I dare to run on a patched NetworkManager, as I am concerned, that Fedora might overwrite changes or becomes incompatible with this patched version over time...

Any additional hints are warmly welcome!

Thank again!
Thomas

essys, 2024-08-15:
I can confirm that this also works on the Lenovo Ideapad Duet 3 (10IGL5) with Fedora 40.

tuxor1337, 2024-08-15:
The latest news is that the ModemManager integration is on its way, and it will work fine with the iosm kernel module in mainline: https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200