xmm7360/xmm7360-pci #225: Please help testing new ModemManager integration

tuxor1337, 2024-06-29:
Based on the Python scripts in this repo, I created a merge request to add support for the XMM7360 in RPC mode to ModemManager: [!1200](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200) If anyone here has capabilities to test my implementation on their setup, please do and report about your experience in the MR. Thanks in advance!

pswiatki, 2024-06-29:
Will do! Tomorrow will be the good time for testing.

d-hancharou, 2024-06-29:
Can you explain how to test it ? Ubuntu 24.04 LTS

tuxor1337, 2024-06-29:
I don't have Ubuntu. But here is how I test it under Fedora 40, and it should be the same under Ubuntu:

1. Install the build dependencies:
```
$ # for Fedora
$ sudo dnf builddep ModemManager
$ 
$ # for Ubuntu
$ sudo apt build-dep modemmanager
```
2. Clone the ModemManager git repository and check out my branch:
```
$ git clone https://gitlab.freedesktop.org/mobile-broadband/ModemManager.git/
$ cd ModemManager
$ git fetch https://gitlab.freedesktop.org/tuxor1337/ModemManager.git/ port-xmm7360
$ git checkout -b 'ModemManager-port-xmm7360' FETCH_HEAD
```
3. Configure and compile ModemManager:
```
$ meson setup build --buildtype=debug --prefix=$PWD/dist/ --sysconfdir=$PWD/dist/etc -Dudevdir=$PWD/dist/lib/udev 
$ ninja -C build install
```
4. Stop and Disable ModemManager with the following commands:
```
$ sudo systemctl stop ModemManager.service
$ sudo systemctl disable ModemManager.service
```
5. Make sure the `iosm` kernel module is loaded. NOTE: You do not need any of the stuff from this GitHub repository. You don't need the xmm7360-pci kernel module, or the Python scripts.
```
$ sudo modprobe iosm
```
6. Run ModemManager in debug mode:
```
$ sudo ./build/src/ModemManager --debug
```
7. Set up a broadband connection with NetworkManager. Disable Wifi (and LAN) check whether you are connected to the internet with your broadband modem, e.g. by opening some websites.

Go to https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200 and report about your experiences. Note that some things are known not to work. See the first post of the merge request for more information about what is and is not expected to work.

pswiatki, 2024-06-30:
> Will do! Tomorrow will be a good time for testing.

Done, works nicely!

pswiatki, 2024-06-30:
> $ export PKG_CONFIG_PATH="$PKG_CONFIG_PATH:$PWD/dist/lib64/pkgconfig"

Note: check the path - it may be slightly different. My build (of libmbim) created ./dist/lib/ (not `./dist/lib64/`)

pswiatki, 2024-06-30:
[image]

kenti714, 2024-07-16:
Super!
It's Wroks!

-----------------------------
$ uname -a
Linux x280 6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
$ sudo dmidecode | grep -A 3 "System Information"
System Information
	Manufacturer: LENOVO
	Product Name: 20KES0PC00
	Version: ThinkPad X280

[image]

stefzg, 2024-07-26:
I would like to share my situation, unfortunately I can't get it to work though.
I'm working on a X1 Carbon Gen 6, Ubuntu 24.04, all firmware / Bios updated. The Fibocom L850-GL works under Windows 11 without issue.
After a clean reboot the Modem is in a 'failed' state with failed-reason "sim-missing".
After installing the patched ModemManager (which appears to work well) I'm no longer getting any Modem in ModemManager. The debug output is this:

`ModemManager[6306]: <msg> [1721982321.580557] [device /sys/devices/pci0000:00/0000:00:1c.2/0000:03:00.0] creating modem with plugin 'Intel' and '4' ports
ModemManager[6306]: <dbg> [1721982321.580622] [plugin/intel] (intel) Intel XMM7360 in RPC mode found...
ModemManager[6306]: <dbg> [1721982321.584788] [modem0] port 'wwan/wwan0xmmrpc0' grabbed
ModemManager[6306]: <wrn> [1721982321.584879] [plugin/intel] could not grab port wwan0at1: Cannot add port 'wwan/wwan0at1', unhandled port type
ModemManager[6306]: <wrn> [1721982321.584986] [plugin/intel] could not grab port wwan0at0: Cannot add port 'wwan/wwan0at0', unhandled port type
ModemManager[6306]: <dbg> [1721982321.585149] [modem0] port 'net/wwan0' grabbed
ModemManager[6306]: <dbg> [1721982321.585249] [modem0] cleaning up port 'wwan/wwan0xmmrpc0'...
ModemManager[6306]: <dbg> [1721982321.585294] [modem0] cleaning up port 'net/wwan0'...
ModemManager[6306]: <dbg> [1721982321.585362] [modem0] completely disposed
ModemManager[6306]: <wrn> [1721982321.585407] [base-manager] couldn't create modem for device '/sys/devices/pci0000:00/0000:00:1c.2/0000:03:00.0': Failed to find primary AT port
ModemManager[6306]: <dbg> [1721982321.585443] [device /sys/devices/pci0000:00/0000:00:1c.2/0000:03:00.0] port probe list reset...
ModemManager[6306]: <dbg> [1721982321.585509] [wwan0at0/at] device open count is 0 (close)
ModemManager[6306]: <dbg> [1721982321.585542] [wwan0at0/at] closing serial port...
ModemManager[6306]: <dbg> [1721982321.591878] [wwan0at0/at] serial port closed
ModemManager[6306]: <dbg> [1721982321.592116] [wwan0at0/at] forced to close port
`

I'm happy to take any pointers / or to try additional steps if I can help this effort.
Kind regards,
Stefan

tuxor1337, 2024-07-26:
Most likely, your modem device is in a frozen state. Make sure to completely disable any older version of ModemManager that you might have on your system. For example `systemctl disable ModemManager.service`. An older version of ModemManager will try to interact with the modem device and might put it into a dead frozen state. Make sure to not only reboot the computer, but really shut down and boot again (cold boot), before trying to use the new version of ModemManager.

d-hancharou, 2024-09-08:
Missing step for ubuntu 24.04:

`sudo apt-get install libglib2.0-dev libgirepository1.0-dev help2man`

GamerClassN7, 2024-10-17:
> meson setup build --buildtype=debug --prefix=$PWD/dist/ --sysconfdir=$PWD/dist/etc -Dudevdir=$PWD/dist/lib/udev

on ubuntu 24 LTS i unforchunately get this error :/

```
ependency qmi-glib found: NO found 1.35.2 but need: '>= 1.35.6'
Did not find CMake 'cmake'
Found CMake: NO
Run-time dependency qmi-glib found: NO 

meson.build:268:17: ERROR: Dependency lookup for qmi-glib with method 'pkgconfig' failed: Invalid version, need 'qmi-glib' ['>= 1.35.6'] found '1.35.2'.

A full log can be found at /tmp/ModemManager/build/meson-logs/meson-log.txt
```

tuxor1337, 2024-10-17:
@GamerClassN7 Yes, in fact, the ModemManager code has changed in the meantime and you now also have to compile the latest development version of `libqmi`. Fortunately, this is very similar to the steps for `libmbim`.

GamerClassN7, 2024-10-17:
> @GamerClassN7 Yes, in fact, the ModemManager code has changed in the meantime and you now also have to compile the latest development version of `libqmi`. Fortunately, this is very similar to the steps for `libmbim`.

I am really not skilled enough, could you please suggest necessary steps ?

tuxor1337, 2024-10-17:
I updated the instructions in my comment above: https://github.com/xmm7360/xmm7360-pci/issues/225#issuecomment-2198104317 Please start with the first step and make sure not to leave out any of the instructions.

GamerClassN7, 2024-10-21:
> I updated the instructions in my comment above: [#225 (comment)](https://github.com/xmm7360/xmm7360-pci/issues/225#issuecomment-2198104317) Please start with the first step and make sure not to leave out any of the instructions.

Thanks ho everi hit another roadblock, `/etc/apt/sources.list`  is empty on ubuntu 24

GamerClassN7, 2024-10-21:
> > I updated the instructions in my comment above: [#225 (comment)](https://github.com/xmm7360/xmm7360-pci/issues/225#issuecomment-2198104317) Please start with the first step and make sure not to leave out any of the instructions.
> 
> Thanks, how ever hit another roadblock, `/etc/apt/sources.list` is empty on ubuntu 24

Hi uddate i found missing peace:
`sed -i 's/^Types: deb$/Types: deb deb-src/' /etc/apt/sources.list.d/ubuntu.sources`

```
sudo ./build/src/ModemManager --debug
ModemManager[17885]: <msg> [1729490381.307107] ModemManager (version 1.23.11) starting in system bus...
ModemManager[17885]: <dbg> [1729490381.307233] [charsets] detecting platform iconv() support...
ModemManager[17885]: <dbg> [1729490381.307317] [charsets]   UTF-8: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.307349] [charsets]   UCS-2BE: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.307373] [charsets]   ASCII: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.307594] [charsets]   ISO8859-1: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.307716] [charsets]   CP437: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.307916] [charsets]   CP850: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.308153] [charsets]   UTF-16BE: iconv conversion to/from charset is supported
ModemManager[17885]: <dbg> [1729490381.309001] Full suspend/resume hooks enabled
ModemManager[17885]: <dbg> [1729490381.309477] [sleep-monitor-systemd] singleton created
ModemManager[17885]: <dbg> [1729490381.311455] bus acquired, creating manager...
ModemManager[17885]: <dbg> [1729490381.314440] [auth-provider] singleton created
ModemManager[17885]: <dbg> [1729490381.314690] [filter] created
ModemManager[17885]: <dbg> [1729490381.314722] [filter]   explicit allowlist:         yes
ModemManager[17885]: <dbg> [1729490381.314737] [filter]   explicit blocklist:         yes
ModemManager[17885]: <dbg> [1729490381.314746] [filter]   plugin allowlist:           yes
ModemManager[17885]: <dbg> [1729490381.314753] [filter]   qrtr devices allowed:       yes
ModemManager[17885]: <dbg> [1729490381.314762] [filter]   virtual devices forbidden:  yes
ModemManager[17885]: <dbg> [1729490381.314773] [filter]   net devices allowed:        yes
ModemManager[17885]: <dbg> [1729490381.314786] [filter]   usbmisc devices allowed:    yes
ModemManager[17885]: <dbg> [1729490381.314806] [filter]   rpmsg devices allowed:      yes
ModemManager[17885]: <dbg> [1729490381.314819] [filter]   wwan devices allowed:       yes
ModemManager[17885]: <dbg> [1729490381.314832] [filter]   tty devices:
ModemManager[17885]: <dbg> [1729490381.314841] [filter]       platform driver check:    yes
ModemManager[17885]: <dbg> [1729490381.314852] [filter]       driver check:             yes
ModemManager[17885]: <dbg> [1729490381.314861] [filter]       cdc-acm interface check:  yes
ModemManager[17885]: <dbg> [1729490381.314873] [filter]       with net check:           yes
ModemManager[17885]: <dbg> [1729490381.314884] [filter]       default:                  forbidden
ModemManager[17885]: <dbg> [1729490381.314972] [plugin-manager] looking for plugins in '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager'
ModemManager[17885]: <dbg> [1729490381.315487] [plugin-manager] loaded shared 'foxconn' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-foxconn.so'
ModemManager[17885]: <dbg> [1729490381.315832] [plugin-manager] loaded shared 'sierra' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-sierra.so'
ModemManager[17885]: <dbg> [1729490381.316169] [plugin-manager] loaded shared 'quectel' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-quectel.so'
ModemManager[17885]: <dbg> [1729490381.316387] [plugin-manager] loaded shared 'option' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-option.so'
ModemManager[17885]: <dbg> [1729490381.316682] [plugin-manager] loaded shared 'telit' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-telit.so'
ModemManager[17885]: <dbg> [1729490381.316899] [plugin-manager] loaded shared 'novatel' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-novatel.so'
ModemManager[17885]: <dbg> [1729490381.317098] [plugin-manager] loaded shared 'mtk' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-mtk.so'
ModemManager[17885]: <dbg> [1729490381.317319] [plugin-manager] loaded shared 'fibocom' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-fibocom.so'
ModemManager[17885]: <dbg> [1729490381.317593] [plugin-manager] loaded shared 'icera' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-icera.so'
ModemManager[17885]: <dbg> [1729490381.317874] [plugin-manager] loaded shared 'xmm' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-xmm.so'
[... cut ...]
^CModemManager[17885]: <msg> [1729490385.527657] caught signal, shutting down...
ModemManager[17885]: <dbg> [1729490385.527751] [base-manager] stopping connection in object manager server
ModemManager[17885]: <msg> [1729490385.529167] ModemManager is shut down
ModemManager[17885]: <dbg> [1729490385.529262] [sleep-monitor-systemd] singleton disposed
ModemManager[17885]: <dbg> [1729490385.529316] [sleep-monitor-systemd] dropping systemd sleep inhibitor
ModemManager[17885]: <dbg> [1729490385.529721] [auth-provider] singleton disposed
```

GamerClassN7, 2024-10-21:
> > > I updated the instructions in my comment above: [#225 (comment)](https://github.com/xmm7360/xmm7360-pci/issues/225#issuecomment-2198104317) Please start with the first step and make sure not to leave out any of the instructions.
> > 
> > 
> > Thanks, how ever hit another roadblock, `/etc/apt/sources.list` is empty on ubuntu 24
> 
> Hi uddate i found missing peace: `sed -i 's/^Types: deb$/Types: deb deb-src/' /etc/apt/sources.list.d/ubuntu.sources`
> 
> also gui is missing the LTE Icon whitch is present in up to date modemManager. nor is asking for pin.
> 
> log below:
> 
> ```
> sudo ./build/src/ModemManager --debug
> ModemManager[17885]: <msg> [1729490381.307107] ModemManager (version 1.23.11) starting in system bus...
> ModemManager[17885]: <dbg> [1729490381.307233] [charsets] detecting platform iconv() support...
> ModemManager[17885]: <dbg> [1729490381.307317] [charsets]   UTF-8: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.307349] [charsets]   UCS-2BE: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.307373] [charsets]   ASCII: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.307594] [charsets]   ISO8859-1: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.307716] [charsets]   CP437: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.307916] [charsets]   CP850: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.308153] [charsets]   UTF-16BE: iconv conversion to/from charset is supported
> ModemManager[17885]: <dbg> [1729490381.309001] Full suspend/resume hooks enabled
> ModemManager[17885]: <dbg> [1729490381.309477] [sleep-monitor-systemd] singleton created
> ModemManager[17885]: <dbg> [1729490381.311455] bus acquired, creating manager...
> ModemManager[17885]: <dbg> [1729490381.314440] [auth-provider] singleton created
> ModemManager[17885]: <dbg> [1729490381.314690] [filter] created
> ModemManager[17885]: <dbg> [1729490381.314722] [filter]   explicit allowlist:         yes
> ModemManager[17885]: <dbg> [1729490381.314737] [filter]   explicit blocklist:         yes
> ModemManager[17885]: <dbg> [1729490381.314746] [filter]   plugin allowlist:           yes
> ModemManager[17885]: <dbg> [1729490381.314753] [filter]   qrtr devices allowed:       yes
> ModemManager[17885]: <dbg> [1729490381.314762] [filter]   virtual devices forbidden:  yes
> ModemManager[17885]: <dbg> [1729490381.314773] [filter]   net devices allowed:        yes
> ModemManager[17885]: <dbg> [1729490381.314786] [filter]   usbmisc devices allowed:    yes
> ModemManager[17885]: <dbg> [1729490381.314806] [filter]   rpmsg devices allowed:      yes
> ModemManager[17885]: <dbg> [1729490381.314819] [filter]   wwan devices allowed:       yes
> ModemManager[17885]: <dbg> [1729490381.314832] [filter]   tty devices:
> ModemManager[17885]: <dbg> [1729490381.314841] [filter]       platform driver check:    yes
> ModemManager[17885]: <dbg> [1729490381.314852] [filter]       driver check:             yes
> ModemManager[17885]: <dbg> [1729490381.314861] [filter]       cdc-acm interface check:  yes
> ModemManager[17885]: <dbg> [1729490381.314873] [filter]       with net check:           yes
> ModemManager[17885]: <dbg> [1729490381.314884] [filter]       default:                  forbidden
> ModemManager[17885]: <dbg> [1729490381.314972] [plugin-manager] looking for plugins in '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager'
> ModemManager[17885]: <dbg> [1729490381.315487] [plugin-manager] loaded shared 'foxconn' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-foxconn.so'
> ModemManager[17885]: <dbg> [1729490381.315832] [plugin-manager] loaded shared 'sierra' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-sierra.so'
> ModemManager[17885]: <dbg> [1729490381.316169] [plugin-manager] loaded shared 'quectel' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-quectel.so'
> ModemManager[17885]: <dbg> [1729490381.316387] [plugin-manager] loaded shared 'option' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-option.so'
> ModemManager[17885]: <dbg> [1729490381.316682] [plugin-manager] loaded shared 'telit' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-telit.so'
> ModemManager[17885]: <dbg> [1729490381.316899] [plugin-manager] loaded shared 'novatel' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-novatel.so'
> ModemManager[17885]: <dbg> [1729490381.317098] [plugin-manager] loaded shared 'mtk' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-mtk.so'
> ModemManager[17885]: <dbg> [1729490381.317319] [plugin-manager] loaded shared 'fibocom' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-fibocom.so'
> ModemManager[17885]: <dbg> [1729490381.317593] [plugin-manager] loaded shared 'icera' utils from '/home/jonatanrek/Desktop/lte3333/ModemManager/dist/lib/x86_64-linux-gnu/ModemManager/libmm-shared-icera.so'
[... cut ...]
> ModemManager[17885]: <dbg> [1729490385.529262] [sleep-monitor-systemd] singleton disposed
> ModemManager[17885]: <dbg> [1729490385.529316] [sleep-monitor-systemd] dropping systemd sleep inhibitor
> ModemManager[17885]: <dbg> [1729490385.529721] [auth-provider] singleton disposed
> ```
> 
> Any idea what could be wrong ?

Interesting for some reason i jeed to togle both wifi and airplane mod to  get it workin after than it is working flawlessly :)  is there vay how to replace lod upstream modem manager with new one ?

bam80, 2024-10-21:
Sorry for jumping in,
it's probably out of topic here,
but I'm not sure where to ask else.

I'm searching for info if L860-GL(16) is supported by MM in USB NCM mode (MBIM over USB doesn't seem implemented on the modem side for IP-link).
Will be gratitude for any support.
Thanks.

GamerClassN7, 2024-10-22:
@tuxor1337 i just noticed possible bug
if computer go to sleeap, and recover from it modem is no where to be found in ui.

[image]

only think that helps is reboot.

pswiatki, 2024-10-22:
Not a bug.

GamerClassN7, 2024-10-22:
> Not a bug.

@pswiatki can i ask what is reason for this behavior ? or how can it be prevented cause from my investigantion is modem manager sucesfully runnin agter wake up.

pswiatki, 2024-10-22:
Wrong modem state after wake up is the reason. It isn't immediately obvious how to bring the modem to a correct state (reInitialize, or something). Read Author's notes to find out more.

GamerClassN7, 2024-10-22:
> Wrong modem state after wake up is the reason. It isn't immediately obvious how to bring the modem to a correct state (reInitialize, or something). Read [Author's notes](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200) to find out more. This issue is explained right at the beginning.

Ou... thhangs of fointing out there are other exelet notes as well :)

jeriox, 2024-11-07:
Hi, thanks so much for your work! Unfortunately I could not get it to work right away. Installation worked flawlessly, but I cannot connect to the mobile network successfully, the NM logs claim that there there is no network service. The modem worked in Windows at the same location just shortly before. Interestingly, coming back to Windows after the installation on Linux, it claims no reception as well. Is this a symptom of the frozen state mentioned earlier? I'll try again at a different location in the next days.

NM logs for reference: 
```
	ModemManager[41257]: <inf> [1731019436.743431] [modem0] processing user request to connect modem...
ModemManager[41257]: <inf> [1731019436.743465] [modem0]   apn: internet.v6.telekom
ModemManager[41257]: <inf> [1731019436.743478] [modem0]   ip type: ipv4
ModemManager[41257]: <inf> [1731019436.743493] [modem0]   allowed auth: none, pap, chap, mschap, mschapv2, eap
ModemManager[41257]: <inf> [1731019436.743509] [modem0]   user: ###
ModemManager[41257]: <inf> [1731019436.743525] [modem0]   password: ###
ModemManager[41257]: <msg> [1731019436.743549] [modem0] simple connect started...
ModemManager[41257]: <msg> [1731019436.743562] [modem0] simple connect state (6/10): register
ModemManager[41257]: <dbg> [1731019436.743596] [modem0] running registration checks (CS: 'yes', PS: 'yes', EPS: 'yes', 5GS: 'no')
ModemManager[41257]: <dbg> [1731019436.743622] [wwan0at1/at] device open count is 2 (open)
ModemManager[41257]: <dbg> [1731019436.743684] [wwan0at1/at] --> 'AT+CREG?<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.745975] [wwan0at1/at] <-- '<CR><LF>+CREG: 2,0<CR><LF><CR><LF>OK<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.746078] [wwan0at1/at] device open count is 3 (open)
ModemManager[41257]: <dbg> [1731019436.746106] [wwan0at1/at] device open count is 2 (close)
ModemManager[41257]: <dbg> [1731019436.746133] [wwan0at1/at] --> 'AT+CGREG?<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.748084] [wwan0at1/at] <-- '<CR><LF>+CGREG: 2,0<CR><LF><CR><LF>OK<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.748162] [wwan0at1/at] device open count is 3 (open)
ModemManager[41257]: <dbg> [1731019436.748190] [wwan0at1/at] device open count is 2 (close)
ModemManager[41257]: <dbg> [1731019436.748216] [wwan0at1/at] --> 'AT+CEREG?<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.750361] [wwan0at1/at] <-- '<CR><LF>+CEREG: 2,0<CR><LF><CR><LF>OK<CR><LF>'
ModemManager[41257]: <inf> [1731019436.750467] [modem0] launching automatic network registration...
ModemManager[41257]: <dbg> [1731019436.750502] [wwan0at1/at] device open count is 3 (open)
ModemManager[41257]: <dbg> [1731019436.750536] [wwan0at1/at] device open count is 2 (close)
ModemManager[41257]: <dbg> [1731019436.750568] [wwan0at1/at] --> 'AT+COPS=0<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.753208] [wwan0at1/at] <-- '<CR><LF>+CME ERROR: 30<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.753305] [wwan0at1/at] operation failure: 30 (No network service)
ModemManager[41257]: <dbg> [1731019436.753373] [modem0] registration in network: timed out
ModemManager[41257]: <wrn> [1731019436.753400] [modem0] registration in network failed: Network timeout
ModemManager[41257]: <dbg> [1731019436.753558] [wwan0at1/at] device open count is 1 (close)
ModemManager[41257]: <inf> [1731019436.771327] [modem0] processing user request to disconnect modem: all bearers
ModemManager[41257]: <inf> [1731019436.771418] [modem0] all requested bearers disconnected
ModemManager[41257]: <inf> [1731019436.772545] [modem0] processing user request to disconnect modem: all bearers
ModemManager[41257]: <inf> [1731019436.772615] [modem0] all requested bearers disconnected
ModemManager[41257]: <inf> [1731019436.773416] [modem0] processing user request to connect modem...
ModemManager[41257]: <inf> [1731019436.773449] [modem0]   apn: internet.v6.telekom
ModemManager[41257]: <inf> [1731019436.773465] [modem0]   ip type: ipv4
ModemManager[41257]: <inf> [1731019436.773480] [modem0]   allowed auth: none, pap, chap, mschap, mschapv2, eap
ModemManager[41257]: <inf> [1731019436.773496] [modem0]   user: ###
ModemManager[41257]: <inf> [1731019436.773513] [modem0]   password: ###
ModemManager[41257]: <msg> [1731019436.773535] [modem0] simple connect started...
ModemManager[41257]: <msg> [1731019436.773551] [modem0] simple connect state (6/10): register
ModemManager[41257]: <dbg> [1731019436.773587] [modem0] running registration checks (CS: 'yes', PS: 'yes', EPS: 'yes', 5GS: 'no')
ModemManager[41257]: <dbg> [1731019436.773620] [wwan0at1/at] device open count is 2 (open)
ModemManager[41257]: <dbg> [1731019436.773682] [wwan0at1/at] --> 'AT+CREG?<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.775958] [wwan0at1/at] <-- '<CR><LF>+CREG: 2,0<CR><LF><CR><LF>OK<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.776057] [wwan0at1/at] device open count is 3 (open)
ModemManager[41257]: <dbg> [1731019436.776089] [wwan0at1/at] device open count is 2 (close)
ModemManager[41257]: <dbg> [1731019436.776122] [wwan0at1/at] --> 'AT+CGREG?<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.778077] [wwan0at1/at] <-- '<CR><LF>+CGREG: 2,0<CR><LF><CR><LF>OK<CR><LF>'
ModemManager[41257]: <dbg> [1731019436.778182] [wwan0at1/at] device open count is 3 (open)
ModemManager[41257]: <dbg> [1731019436.778215] [wwan0at1/at] device open count is 2 (close)
ModemManager[41257]: <dbg> [1731019436.778273] [wwan0at1/at] --> 'AT+CEREG?<CR><LF>'
[... cut ...]
ModemManager[41257]: <dbg> [1731019459.017697] [modem0] removing 3GPP +CGEV unsolicited events handlers in wwan0at1
ModemManager[41257]: <dbg> [1731019459.017720] [modem0] disabling +CGEV event reporting in primary port...
ModemManager[41257]: <dbg> [1731019459.017738] [wwan0at1/at] device open count is 2 (open)
ModemManager[41257]: <dbg> [1731019459.017764] [wwan0at1/at] --> 'AT+CGEREP=0<CR><LF>'
ModemManager[41257]: <dbg> [1731019459.019362] [wwan0at1/at] <-- '<CR><LF>OK<CR><LF>'
ModemManager[41257]: <dbg> [1731019459.019390] [modem0] disabling the Modem interface...
```

michael-dev, 2025-02-19:
I've been testing intel-rpc-xmm7360-dan at 3f29af1266. I've seen the same issue as jeriox, but it went away after booting into windows once again and then rebooting into linux and I got internet access.

I'm now seeing random kernel hangups (sometimes no log, sometimes in seemingly unrelated areas, e.g. swap). I'm using Ubuntu 24.10 at Kernel 6.11.0-14-generic #15-Ubuntu x86_64. So maybe something is not yet right.

hsiaov, 2025-03-03:
encounter complie ModemManager after try in 24.04, any further recommandation?

meson setup build --buildtype=debug --prefix=$PWD/dist/ --sysconfdir=$PWD/dist/etc -Dudevdir=$PWD/dist/lib/udev
The Meson build system
Version: 1.3.2
Source dir: /home/GitHub/xmm7360-pci/ModemManager
Build dir: /home/GitHub/xmm7360-pci/ModemManager/build
Build type: native build
Project name: ModemManager
Project version: 1.23.12
C compiler for the host machine: cc (gcc 13.3.0 "cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
C linker for the host machine: cc ld.bfd 2.42
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program python3 found: YES (/usr/bin/python3)
Program /home/vincent/文件/GitHub/xmm7360-pci/ModemManager/build-aux/mm-mkenums found: YES (/home/vincent/文件/GitHub/xmm7360-pci/ModemManager/build-aux/mm-mkenums)
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Wno-cast-function-type: YES 
Compiler for C supports arguments -Wno-packed: YES 
Compiler for C supports arguments -Wno-float-equal: YES 
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -fno-strict-aliasing: YES 
Compiler for C supports arguments -Waggregate-return: YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wdouble-promotion: YES 
Compiler for C supports arguments -Wduplicated-branches: YES 
Compiler for C supports arguments -Wduplicated-cond: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Winline: YES 
Compiler for C supports arguments -Wjump-misses-init: YES 
Compiler for C supports arguments -Wlogical-op: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wmaybe-uninitialized: YES 
Compiler for C supports arguments -Wmissing-declarations: YES 
Compiler for C supports arguments -Wmissing-format-attribute: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-noreturn: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wnull-dereference: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wredundant-decls: YES 
Compiler for C supports arguments -Wrestrict: YES 
Compiler for C supports arguments -Wreturn-type: YES 
Compiler for C supports arguments -Wshadow: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wsuggest-attribute=format: YES 
Compiler for C supports arguments -Wswitch-default: YES 
Compiler for C supports arguments -Wswitch-enum: YES 
Compiler for C supports arguments -Wundef: YES 
Compiler for C supports arguments -Wunused-but-set-variable: YES 
Compiler for C supports arguments -Wwrite-strings: YES 
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Run-time dependency gio-unix-2.0 found: YES 2.80.0
Run-time dependency glib-2.0 found: YES 2.80.0
Run-time dependency gmodule-2.0 found: YES 2.80.0
Run-time dependency gio-2.0 found: YES 2.80.0
Run-time dependency gobject-2.0 found: YES 2.80.0
Run-time dependency dbus-1 found: YES 1.14.10
Run-time dependency bash-completion found: YES 2.11
Run-time dependency gudev-1.0 found: YES 238
Run-time dependency systemd found: YES 255
Run-time dependency libsystemd found: YES 255
Run-time dependency polkit-gobject-1 found: YES 124
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 194, in run
    return options.run_func(options)
[... 76 lines cut ...]
    ret, out, err = self._call_pkgbin(args, env=env)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/mesonbuild/dependencies/pkgconfig.py", line 280, in _call_pkgbin
    p, out, err = Popen_safe_logged(cmd, env=env)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/mesonbuild/utils/universal.py", line 1572, in Popen_safe_logged
    p, o, e = Popen_safe(args, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/mesonbuild/utils/universal.py", line 1530, in Popen_safe
    o, e = p.communicate(write)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1209, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 2153, in _communicate
    stdout = self._translate_newlines(stdout,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1086, in _translate_newlines
    data = data.decode(encoding, errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 17: invalid continuation byte

meson.build:261:18: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!

pswiatki, 2025-03-03:
If I may ask: what is this?
```
 /home/vincent/文件/GitHub/xmm7360-p.....
               ^^^
```

kenti714, 2025-05-16:
> May I ask what this is?
> 
> ```
>  /home/vincent/文件/GitHub/xmm7360-p.....
>                ^^^
> ```

May be Chinese for "Documents"

BTW,When will the XMM7360 officially work with ModemManager?
It's hard to carry a PC without sleepmode. ;-)

pswiatki, 2025-05-16:
Use this modem with `iosm` driver. I believe the issue with sleep has been fixed and the RPC support is operational (no functionality missing as far as I can tell). Check the title here (top of page). Then check [this Merge Request](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1280) in `Modem Manager` GitLab.

pswiatki, 2025-05-16:
> May be Chinese for "Documents"

Well.... I was initially wondering if this can have anything to do with the build failure (exotic characters in the path).

SimPilotAdamT, 2025-05-18:
Just a quick question, I've noticed that the upstream MR [!1200](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200) was closed without being merged. I take it this will be re-created pending further testing?

If that's the case, I'm happy to help test using my ThinkPad P52s

tuxor1337, 2025-05-18:
> Just a quick question, I've noticed that the upstream MR [!1200](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200) was closed without being merged. I take it this [...]

The work of the above-mentioned merge request has been finalized in https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1235 The result will be available in upcoming ModemManager releases. However, some features (esp. regarding stability) are still work in progress, see https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1280

closed 2025-05-18 by tuxor1337

Peter-Ries, 2025-12-25:
@tuxor1337  - first of all thanks for your efforts to get this Modem bein supported by ModemManager.

I somehow ended up here following endless threads in the web.

I have an AMD Gen1 Thinkpad T14 with the Fibocom L850-GL LTE and a German Telekom SIMcard (no PIN). All running on Endeavour OS (Arch based and latest GNOME 49. I already got this card to work two or more years ago with the xmm7360 project and the setup-lte.sh script in there, but it stopped working. So here another try,

I was albe to compile and install the ModemManager with the help of your description above (may be you could remove the "$" from the commands to run as copy&paste needs editing). So here is the stuff I did for Arch based distributions. Your documentation needs to be adjusted where yo already have the Fedora/Ubuntu part in.

```bash
# +++ add this start
# packages needed for ARCH based distros 
sudo pacman -S base-devel modemmanager libmbim libqmi help2man
# +++ end

# just my working directory
mkdir -p /Git/modemmanager-git
cd ~/Git/modemmanager-git

# cloning, fetching, etc...
git clone https://gitlab.freedesktop.org/mobile-broadband/ModemManager.git/
git clone https://gitlab.freedesktop.org/mobile-broadband/libmbim.git/
git clone https://gitlab.freedesktop.org/mobile-broadband/libqmi.git/
cd ModemManager
git fetch https://gitlab.freedesktop.org/tuxor1337/ModemManager.git/ port-xmm7360
git checkout -b 'ModemManager-port-xmm7360' FETCH_HEAD

cd ../libmbim
meson setup build --buildtype=debug --prefix=$PWD/dist/ 
ninja -C build install

# +++ add this start
# --- EXPORTS for ARCH based distro
export PKG_CONFIG_PATH="$PKG_CONFIG_PATH:$PWD/dist/lib/pkgconfig"
export CFLAGS="-I$PWD/dist/include/libmbim-glib"
export LDFLAGS="-L$PWD/dist/lib"
# +++ end

cd ../libqmi
meson setup build --buildtype=debug --prefix=$PWD/dist/ 
ninja -C build install

# +++ add this start
# --- EXPORTS for ARCH based distro
export PKG_CONFIG_PATH="$PKG_CONFIG_PATH:$PWD/dist/lib/pkgconfig"
export CFLAGS="-I$PWD/dist/include/libqmi-glib"
export LDFLAGS="-L$PWD/dist/lib"
# +++ end

# use this part to get stuff up & running
# vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
cd ../ModemManager
meson setup build --buildtype=debug --prefix=$PWD/dist/ --sysconfdir=$PWD/dist/etc -Dudevdir=$PWD/dist/lib/udev 
ninja -C build install

sudo chown root:root ./dist/share/ModemManager/fcc-unlock.available.d/8086:7360
sudo ln -s $PWD/dist/share/ModemManager/fcc-unlock.available.d/8086:7360 ./dist/etc/ModemManager/fcc-unlock.d/
sudo ln -s $PWD/dist/share/ModemManager/fcc-unlock.available.d/8086:7360 ./dist/lib64/ModemManager/fcc-unlock.d/

sudo systemctl stop ModemManager.service
sudo systemctl disable ModemManager.service

sudo modprobe iosm

sudo ./build/src/ModemManager --debug
``` 

After I ran the script the Mobils Broadband appeared in GNOME-Settings. It came with a "Prepaid Mobile" for "Congstar" APN. I tried this one and also tried a modification with Telekom settings.

What always happened was that AFTER activating the Mobile Broadband the modemmanager script terminated and the setting disappeared from Settings dialog. The whole entry was gone, just WiFi, Network and Bluetooth left.

For another try I needed to completely poweroff the laptop and cold boot. Simple reboot was not enough.

I attached the whole debug log of the test to this posting. Hope it helps.

[modemmanager_error.log](https://github.com/user-attachments/files/24339384/modemmanager_error.log)

maciejsszmigiero, 2025-12-25:
@Peter-Ries your modem is FCC locked and the unlock script executed by ModemManager is apparently timing out for some reason:
> ModemManager[6198]: <dbg> [1766675130.167967] [wwan0xmmrpc0/xmmrpc] serial port closed
> ModemManager[6198]: <wrn> [1766675135.567159] [fcc unlock dispatcher] forcing exit on fcc unlock operation

maciejsszmigiero, 2025-12-26:
Created a [Modem Manager task](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/1034) about fixing these ModemManager asserts that @Peter-Ries is hitting.

This is unlikely to fix the FCC unlock issue, however.

pswiatki, 2025-12-27:
> [@Peter-Ries](https://github.com/Peter-Ries) Retried the same on my system (also Arch, pure, Dell Precision 3561) and it still worked very well.

By the way, location service is also operational, and GPS provides correct coordinates.
So, all good! (still)
The only thing that would be nice to have working would be proper modem initialisation after Suspend / Hibernate. 
But that one is really hard, as far as I know.

Peter-Ries, 2025-12-27:
thanks for your updates guys - appreciate it very much. 

I discovered a typo in the symlinks of my script. Fixed this in my [posting](https://github.com/xmm7360/xmm7360-pci/issues/225#issuecomment-3691556822) and will now read through your comments.

Peter-Ries, 2025-12-27:
Unfortunately I could not yet get around my eror. Gemini and chatgpt suggested fixes to code, fcc unlock stuff via screen command and AT commends directly, udev rules, a.s.o. - I somehow drifted away from the real solution I guess, so I gave up for the moment. Will start with a fresh head the next days. May be it's all a frmware thing?!

But thanks so far to all thet are working on getting this card to operate!

arf20, 2025-12-27:
> [@Peter-Ries](https://github.com/Peter-Ries) Retried the same on my system (also Arch, pure, Dell Precision 3561) and it still worked very well:
> <img alt="Image" width="1521" height="691" src="https://private-user-images.githubusercontent.com/7026139/530438011-f0ad8435-5092-4e02-9e2c-bbd9f5a81a1e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjY4NDk5MDksIm5iZiI6MTc2Njg0OTYwOSwicGF0aCI6Ii83MDI2MTM5LzUzMDQzODAxMS1mMGFkODQzNS01MDkyLTRlMDItOWUyYy1iYmQ5ZjVhODFhMWUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTIyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEyMjdUMTUzMzI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWQ2MzZjY2EwMTBkZTk3MzVjNmNmYTQ5ZTU1YjRmYjBkNzlmOTUyYzQ2MWUyZWE1ZjY4NDNkNDc5MzIwZjgzYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.dywW_mMJWE-w-pYgPBEplD7DuSz6tIVAfEx4Y49s2iU">
> 
> This result might also be of interest to [@arf20](https://github.com/arf20) The only difference in the procedure was the `--reconfigure` switch in `meson` - reason being I had the old, already used, source trees already in place, so I had to add the switch to reconfigure the build (just in case).

I still cant get it to do anything on my Debian 13 Thinkpad P53.
```
ModemManager[866052]: <dbg> [1766849583.998112] [base-manager] adding port wwan0xmmrpc0 at sysfs path: /sys/devices/pci0000:00/0000:00:1c.6/0000:53:00.0/wwan/wwan0/wwan0xmmrpc0
ModemManager[866052]: <dbg> [1766849583.998117] [base-manager] port wwan0xmmrpc0 not candidate
```
Now I tried compiling the `ModemManager-port-xmm7360` branch to no avail.

maciejsszmigiero, 2025-12-27:
> git fetch https://gitlab.freedesktop.org/tuxor1337/ModemManager.git/ port-xmm7360
> git checkout -b 'ModemManager-port-xmm7360' FETCH_HEAD

That's not longer current - please use the `main` branch of the [upstream ModemManager repository](https://gitlab.freedesktop.org/mobile-broadband/ModemManager) instead since XMM7360 support was merged there months ago.

It's also the upstream ModemManager repository that got and will get all further fixes.

arf20, 2025-12-27:
> That's not longer current - please use the `main` branch of the [upstream ModemManager repository](https://gitlab.freedesktop.org/mobile-broadband/ModemManager) instead since XMM7360 support was [...]

yeah I tried both, but I do not know what is wrong with my card or ModemManager, it refuses to mark it as candidate

Peter-Ries, 2025-12-28:
@pswiatki - that was the line I could not find. 
Chatgpt suggested other fixes to code that didn't work. Will try this one line fix in the next days

Many thanks!!! 👍

pswiatki, 2025-12-28:
It seems, on my system (Dell Precision 3561) it can be rectified by the following sequence of commands:

```
17859  2025-12-28 11:25:21 rmmod iosm
17860  2025-12-28 11:25:35 echo "1" > /sys/bus/pci/devices/0000\:73\:00.0/reset 
17861  2025-12-28 11:26:05 modprobe -v iosm
```
MM picks up the emergence of the new modem device and configures it correctly.

tuxor1337, 2025-12-28:
Thanks everyone for your continued interest in this implementation! And special thanks to everyone who is trying to give support to people that ask for help in this thread! :) Generally, if you run into problems with the ModemManager implementation, please consider opening a ticket in the ModemManager repository instead of asking me for help via this thread.

> Now I need to wait until this version (1.25.95) is being released through Fedora's repository (which currently holds v1.24.2)...

Note that 1.25.95 is a testing release. We will have to wait for 1.26 for a stable release that includes XMM7360 support: https://lists.freedesktop.org/archives/modemmanager-devel/2025-December/010076.html

@pswiatki @Peter-Ries Note that the FCC-unlock is not done in the C code, but in a separate bash script (https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/blob/main/data/dispatcher-fcc-unlock/8086:7360). Hence, changing the salt in the C code will not fix your issue. You need to change the VENDOR_ID_HASH in the bash script. This is because of a ModemManager policy that all FCC unlock operations need to be optional and disabled by default: https://modemmanager.org/docs/modemmanager/fcc-unlock/

pswiatki, 2025-12-29:
Wrong patch (irrelevant), as @tuxor1337 explained. Try this, perhaps it will work better for you:
```
--- a/data/dispatcher-fcc-unlock/8086:7360
+++ b/data/dispatcher-fcc-unlock/8086:7360
@@ -78,7 +78,10 @@ rpc_command() {
     exec 99>&-
 }
 
-VENDOR_ID_HASH="3df8c719"
+
+# Try null vendor hash (salt), as per https://github.com/xmm7360/xmm7360-pci/issues/240
+#VENDOR_ID_HASH="3df8c719"
+VENDOR_ID_HASH="00000000"
 
 for i in {1..3}; do
     log "Attempt #${i} to unlock WWAN modem"
```
Good luck!

pswiatki, 2025-12-29:
At least we are getting somewhere (no crash). Perhaps @maciejsszmigiero would be able to tell why your device behaves like this now.

pswiatki, 2025-12-29:
> [@pswiatki](https://github.com/pswiatki):
> 
> > edit: ohh.... I may have forgotten - perhaps hibernation was OK even before, but definitely suspend is not:
> 
> As far as I remember, I have already provided a workaround patch for the s2idle `iosm` issue on your platform [here](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/992#note_2938256) in the past.

Yes, I recall you did, Macieju @maciejsszmigiero. Thank you so much.

Peter-Ries, 2025-12-29:
regarding my issue:  

to be clear I do not use the xmm7360-pci module from here... Out of curiosity I just tried to make / install it but it gives me a make error. Then i tried to use the package from AUR (xmm7360-pci-spat), but this installed a module only for 6.12 LTS kernel, my current 6.18 it did not install neither.

So I'm clueless what needs to be done in addition to compiling this ModemManager as described. I read that curent iosm has it all to operate this modem and just this ModemManager port is needed.

Peter-Ries, 2025-12-29:
tried another fresh recompile with preceeding cold boot. Same error.

I'll swap the SIM card of my x270 with this t14 one, just to cross check.

Somehow I have no other idea than a defect SIM.

Will report back,

pswiatki, 2025-12-30:
Maybe a bad SIM slot?

maciejsszmigiero, 2025-12-30:
I think that these modems report no SIM card via AT command interface if not properly initialized via XMMRPC.

If these ChatGPT-suggested changes resulted in skipping most of XMMRPC initialization then that's kind-of expected result.

pswiatki, 2025-12-30:
> I already reported that, but then it turned out to use iosm and ModemManager - that's where I am right now ;)

You have mix of both and it is a Total Mess, right?
Would you like me to try to build  `xmm7360-pci` for you?

Peter-Ries, 2025-12-31:
Compiling from sources with make somehow throws errors here. That's why I installed the 7360 dkms from AUR. Installation worked, but setting things up with the sudo lte setup resulted in the error I also reported [on package site](https://aur.archlinux.org/packages/xmm7360-pci-spat-dkms-git) - basically same error as posted above.

If you could try to build it from sources, just to check...(And execute the lte script may be?)

No clue why it does not work here. I should have all prerequisites: Kernel 6.18.x plus headers, base-devel help2man gobject-introspection glib2-devel, plus the two python-pyroute and python-configargparse packages.

pswiatki, 2026-01-01:
But wait! Isn't ` xxd ` a hex dumper? If this is what we are talking about, then I don't understand.

maciejsszmigiero, 2026-01-01:
That's a good finding @timothyhoward,  thanks.

It would be nice if @Peter-Ries could confirm that installing `xxd` fixes the FCC unlock issue for him too.
Just make sure to use the upstream ModemManager repo.

Peter-Ries, 2026-01-04:
Just for the docs: In **ARCH** there's a standalone package called `tinyxxd`. So no need to install vim. So just `sudo -S tinyxxd`!