xmm7360/xmm7360-pci #226: Ubuntu 24.04 LTS kernel 6.8.0-36 compilation ERROR

d-hancharou, 2024-06-29:
`make && make load`
```
make -C /lib/modules/6.8.0-36-generic/build M=/home/pret/tmp/v3/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-36-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/pret/tmp/v3/xmm7360-pci/xmm7360.o
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:656:5: warning: no previous prototype for ‘xmm7360_cdev_open’ [-Wmissing-prototypes]
  656 | int xmm7360_cdev_open(struct inode *inode, struct file *file)
      |     ^~~~~~~~~~~~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:664:5: warning: no previous prototype for ‘xmm7360_cdev_release’ [-Wmissing-prototypes]
  664 | int xmm7360_cdev_release(struct inode *inode, struct file *file)
      |     ^~~~~~~~~~~~~~~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:670:9: warning: no previous prototype for ‘xmm7360_cdev_write’ [-Wmissing-prototypes]
  670 | ssize_t xmm7360_cdev_write(struct file *file, const char __user *buf,
      |         ^~~~~~~~~~~~~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:684:9: warning: no previous prototype for ‘xmm7360_cdev_read’ [-Wmissing-prototypes]
  684 | ssize_t xmm7360_cdev_read(struct file *file, char __user *buf, size_t size,
      |         ^~~~~~~~~~~~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c: In function ‘xmm7360_irq0’:
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:1191:13: warning: the comparison will always evaluate as ‘true’ for the address of ‘td_ring’ will never be NULL [-Waddress]
 1191 |         if (xmm->td_ring) {
      |             ^~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:225:24: note: ‘td_ring’ declared here
  225 |         struct td_ring td_ring[16];
      |                        ^~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c: At top level:
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:1342:18: error: initialization of ‘ssize_t (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘long int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct tty_struct *, const unsigned char *, int)’ [-Werror=incompatible-pointer-types]
 1342 |         .write = xmm7360_tty_write,
      |                  ^~~~~~~~~~~~~~~~~
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:1342:18: note: (near initialization for ‘xmm7360_tty_ops.write’)
/home/pret/tmp/v3/xmm7360-pci/xmm7360.c:1445:6: warning: no previous prototype for ‘xmm7360_dev_init_work’ [-Wmissing-prototypes]
 1445 | void xmm7360_dev_init_work(struct work_struct *work)
      |      ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/pret/tmp/v3/xmm7360-pci/xmm7360.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-36-generic/Makefile:1926: /home/pret/tmp/v3/xmm7360-pci] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-36-generic'
make: *** [Makefile:9: default] Error 2
```

sceddd, 2024-07-15:
Have you got any idea yet?

pswiatki, 2024-07-15:
Use `iosm` with the startup scripts here - it will be easier. Or check out `Modem Manager` patch which can now connect and disconnect nicely. That will make things even easier.

impactevent, 2024-08-18:
@pswiatki Are you able to write something more? How to do it?

pswiatki, 2024-08-18:
Yes. I will post a link to the appropriate MM branch

pswiatki, 2024-08-19:
Here is the [link to MM branch](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200):

impactevent, 2024-08-19:
Not Working :(
Point 4
Last step `ninja -C build install`

```
../src/plugins/quectel/mm-port-mbim-quectel.c: In function ‘iface_port_at_check_support’:
../src/plugins/quectel/mm-port-mbim-quectel.c:64:84: error: ‘MBIM_CID_QDU_COMMAND’ undeclared (first use in this function)
```

pswiatki, 2024-08-19:
Use [the second description further down](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1200#note_2473678). Thank you for your understanding.
For reference;

impactevent, 2024-08-20:
@pswiatki thank you, but what second description ? ;)

pswiatki, 2024-08-20:
Didn't I put a link directly to it? If you can't reach it - read the thread carefully and you will know all you need to know. Good luck!

impactevent, 2024-08-20:
@pswiatki you do, and it looks like this branch is not working, can't build on it, still have the same error ..

```
../src/plugins/quectel/mm-port-mbim-quectel.c: In function ‘iface_port_at_check_support’:
../src/plugins/quectel/mm-port-mbim-quectel.c:64:84: error: ‘MBIM_CID_QDU_COMMAND’ undeclared (first use in this function)
```

pswiatki, 2024-08-20:
I did it on ARCH Linux, but was lucky (?) enough to manage everything nicely. Just followed the instructions and explanations given by Thomas (The Author). I never used Ubuntu with `xmm7360` device. I am sure you will be able to succeed - just follow everything precisely and understand what is needed to make it work. Good luck!

lxs602, 2024-12-19:
> `make && make load`
>
> ```
> make -C /lib/modules/6.8.0-36-generic/build M=/home/pret/tmp/v3/xmm7360-pci modules
> make[1]: Entering directory '/usr/src/linux-headers-6.8.0-36-generic'
> warning: the [...]

Changing line 1281 of xmm7360.c from:
``` static int xmm7360_tty_write(struct tty_struct *tty, 
                             const unsigned char *buffer, int count)```

to:
``` static ssize_t xmm7360_tty_write(struct tty_struct *tty,
                             const u8 *buffer, size_t count) ```

and then:
```make CFLAGS="-Wno-error"```

allowed me to make on Ubuntu 24.04 6.8.0-45.

Acknowledgements: ChatGPT (sadly)