xmm7360/xmm7360-pci #229: Thinkpad T15 /Fibocom L850 /Ubuntu 20.04 - doesn't work, doesn't see modem

roma-ar, 2024-08-24:
Please help
In windows 10 work fine but I need to work in ubuntu. Sim is without pin code.
[image]

I tried to install but it was with errors.
----------------------------------------------------------------------------------------------------------------
verifying:

root@ThinkPad-T15-Gen-1:~/tmp/xmm7360-pci# lspci
2c:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)

root@ThinkPad-T15-Gen-1:~# sudo mmcli --list-modems
No modems were found
-----------------------------------------------------------------------------------------------------------------
Install errors:

root@ThinkPad-T15-Gen-1:~/xmm7360-pci-master# make load
sudo /sbin/rmmod iosm
rmmod: ERROR: Module iosm is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:19: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Invalid parameters
make: *** [Makefile:20: load] Error 1

--------------------

root@ThinkPad-T15-Gen-1:~/xmm7360-pci-master# sudo ./scripts/lte.sh setup
lte.sh: manage xmm7360-pci
APN: internet.unite.md
Script: /root/xmm7360-pci-master/scripts/lte.sh; Link: /usr/local/bin

Requirement already satisfied: pyroute2 in /usr/lib/python3/dist-packages (0.5.9)
Requirement already satisfied: configargparse in /usr/lib/python3/dist-packages (0.13.0)
make -C /lib/modules/5.15.0-119-generic/build M=/root/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-119-generic'
  CC [M]  /root/xmm7360-pci-master/xmm7360.o
  MODPOST /root/xmm7360-pci-master/Module.symvers
  CC [M]  /root/xmm7360-pci-master/xmm7360.mod.o
  LD [M]  /root/xmm7360-pci-master/xmm7360.ko
  BTF [M] /root/xmm7360-pci-master/xmm7360.ko
Skipping BTF generation for /root/xmm7360-pci-master/xmm7360.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-119-generic'
sudo /sbin/rmmod iosm
rmmod: ERROR: Module iosm is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:19: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko

-------------------

root@ThinkPad-T15-Gen-1:~/xmm7360-pci-master# lte up
lte.sh: manage xmm7360-pci
APN: internet.unite.md
Script: /root/xmm7360-pci-master/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
[... 154 lines cut ...]
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x5, 0x103, 0x0, 0x1, 0x66, 0x0, 0x67, 0x3, 0x0, 0x0, 0xff, 0x0, 0x0, 0x0, 0x0, 0x0, 0xffff, 0x9d1700, 0x0, 0x5, 0x103, 0x0
response: 0x0, 0x0, 0x0, 0x50103, 0x0, 0x0, 0x0, 0x9, 0x67, 0x3, 0x0, 0x5, 0x103, 0x0, 0x0
RPC executing UtaMsCallPsGetNegIpAddrReq
b'2500000002040000002502040000004911000101020411000101020100020400000000020400000000'
unsolicited: UtaMsNetRegistrationInfoIndCb: 0x0, 0x0, 0x0, 0x0, 0x5, 0x103, 0x0, 0x63, 0x103, 0x0, 0x5, 0x103, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x0, 0x1, 0x66, 0x0, 0x5, 0x103, 0x0, 0x0, 0xff, 0xffff, 0xffffffff, 0xffff, 0xffff, 0x1, 0x0, 0x5, 0x103, 0x0, 0x0, 0x67, 0x3, 0x0, 0x5, 0x103, 0x0, 0x0, 0xff, 0x11, 0x0, 0xff, 0x11, 0x9, 0x0
unsolicited: UtaMsNetRegisteredPlmnNameIndCb: 0x0, bytearray(b'\x06\x8dUg\x92Z\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, 0x5, 0x103, 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0xff, 0x1, 0x0
async_ack: 0x0
response: 0x0, bytearray(b'd`\xbe\x9e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x4, 0x0, 0x0, 0x0
RPC executing UtaMsCallPsGetNegotiatedDnsReq
b'2500000002040000002502040000004711000101020411000101020100020400000000020400000000'
async_ack: 0x0
response: 0x0, bytearray(b'\xac\x1a\x87\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x1, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x0, 0x1, bytearray(b'\xac\x1a\x86\x14\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x1, 0x0, 0x0, 0x0
INFO:root:IP address: 100.96.190.158
INFO:root:DNS server(s): 172.26.135.24
RPC executing UtaMsCallPsConnectReq
b'2b00000002040000002b02040000005111000101020411000101020100020400000006020400000000020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0xffffffff, 0x32, 0x1, 0x3, 0x5, 0x0, 0x9e, 0x0, 0xde, 0x0, 0x0, 0xff, 0xff, 0x3, 0x12, 0x4, 0x7, 0xfe, 0xf [... cut ...]
b'4601000002040000014602040000007d11000100020100020400000000020400000000020400000000020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020100020400000000551402040000001702040000000300000000000000000000000000000000000000000000000204ffffffff02040000003202040000000102040000000302040000000502010002019e0201000201de0201000201000201ff0201ff0201030201120201040201070201fe0201fe02019602010102010202010402010102011f02010902010202010302010402010502040000000155140204000000160204000000026460be9e0000000000000000000000000000000000000204000000000204000000000204ffffffff020400000000'
response: 0xffffffff

pswiatki, 2024-08-24:
Can it be a mismatch between the kernel version against which the driver was built vs. the kernel in use?

Also: perhaps it makes more sense to use `iosm` driver and Thomas' branch of Modem Manager that supports `xmm7360` very nicely?

roma-ar, 2024-08-24:
I reinstaled ubuntu 20.04.6 and tried :
 Compiling xmm7360.ko with DKMS

results:

root@ThinkPad-T15-Gen-1:/home/admin2/xmm7360-pci-master# sudo apt install build-essential python3-pyroute2 python3-configargparse git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-configargparse is already the newest version (0.13.0-2).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.13).
python3-pyroute2 is already the newest version (0.5.9-0ubuntu2).
0 to upgrade, 0 to newly install, 0 to remove and 4 not to upgrade.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up acpi-call-dkms (1.1.0-5) ...
Removing old acpi-call-1.1.0 DKMS files...

------------------------------
Deleting module version: 1.1.0
completely from the DKMS tree.
------------------------------
Done.
Loading new acpi-call-1.1.0 DKMS files...
Building for 5.15.0-119-generic
Building initial module for 5.15.0-119-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/acpi-call-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.15.0-119-generic (x86_64)
Consult /var/lib/dkms/acpi-call/1.1.0/build/make.log for more information.
dpkg: error processing package acpi-call-dkms (--configure):
 installed acpi-call-dkms package post-installation script subprocess returned error exit status 
10
Errors were encountered while processing:
 acpi-call-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

roma-ar, 2024-08-24:
> Can it be a mismatch between the kernel version against which the driver was built vs. the kernel in use?
> 
> Also: perhaps it makes more sense to use `iosm` driver and Thomas' branch of Modem Manager that supports `xmm7360-pci` very nicely?

Hi give me a link please

pswiatki, 2024-08-24:
> 
> Hi give me a link please
> 
OK. Be patient, I shall come beck with information soon.