xmm7360/xmm7360-pci #235: Failed on Lenovo X1 Carbon Gen7

tsoua2, 2025-05-19:
OS: Arch linux  6.14.6-zen1-1-zen
Packages installed: 
  xmm7360-pci-spat-dkms-git 1:0.r252.gdb2bba5-1<SUP>AUR</SUP>
  xmm7360-pci-spat-utils-git 1:0.r261.g27c1b26-1<SUP>AUR</SUP>

Interface tested works on Windows 11, PIN removed.

can see wwan0 interface after boot up
```
6: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none
    inet 25.111.236.24/32 scope global wwan0
       valid_lft forever preferred_lft forever
```

but can't see interface while creating mobile boardband connection in use of NetworkManager

pswiatki, 2025-05-19:
Perhaps the best course of action is to embrace the RPC translation for `iosm` driver and live in peace ever after.

tsoua2, 2025-05-20:
> Perhaps the best course of action is to embrace the RPC translation for `iosm` driver and live in peace ever after. Check this [merge request](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1280) and related MRs (mentioned by Maciej).

Thanks for sharing

closed 2025-05-29 by tsoua2